kecinzer/hpelitebook850g5-opencore #1: How to

Rixxx-git, 2020-03-05:
Thank you for sharing your work, @kecinzer !

On which MacOS Version do you run this and what is better about it than clover? :-)

Joaotcs, 2020-03-10:
I tried to use this. Cannot boot. It stays black with some errors.. (the last one being about "vault" not being configured).

kecinzer, 2020-03-16:
@Rixxx-git 
I just simply removed my EFI partition content and replaced with this. I run it with macOS 10.15.3.
Currently there is no major pros against Clover, its just newer.

@Joaotcs
Please try to use first commit with version 0.5.5. I didn't tested latest version.

Sylvain08, 2020-03-18:
Hi kecinzer,
I'm trying to make my very first Hackingtosh on a 840G5.
So i have a USB key with 32Gb, splited in two partitions.
One with the "install MacOs Catalina" file and one with the EFI folder...
I droped your repo in the EFI partition.
Change the serial number in the config.plist file with the GenSMBIOS script.
But OC doesn't start. I have a black screen with these messages :

`OCS: No Schema for UsePicker at 5 index!`
`OCS: No Schema for RequireSignature at 4 index!`
`OCS: No Schema for RequireVault at 5 index!`
`OCS: No Schema for ConsoleControl at 6 index!`
`OCS: No Schema for AvoidHighAlloc at 0 index!`
`OCS: No Schema for ClearScreenOnModeSwitch at 1 index!`
`OCS: No Schema for ProvideConsoleGop at 5 index!`
`OCS: No Schema for ReconnectOnResChange at 6 index!`
`OCS: No Schema for ReplaceTabWithSpace at 8 index!`
`OCS: No Schema for SanitiseClearScreen at 11 index!`
`OC: Configuration requires vault but no vault provided!`

Maybe I should precise that I do all of stuff on a virtual machine.

Any ideas ?

closed 2020-04-20 by kecinzer