kecinzer/hpelitebook850g5-opencore #5: Start WIN10 restart

wdrich, 2020-04-04:
Start BOOTCAMP partition WIN10 system restarts repeatedly.
Looking forward to your reply!

kecinzer, 2020-04-09:
I'm so sorry, but I don't use Windows at all.

Joaotcs, 2020-04-17:
You should use the boot.efi from windows (using boot menu when your laptop starts). If you enter windows from opencore, it won't work.

kasik96, 2020-04-20:
Using BootCamp with Hackintosh is not good practice. You should install Win 10 after that Hackintosh and OpenCore will take care of rest after booting. If you didnt really mean BootCamp, and just Dual Boot, maybe your EFI in Windows folder is damaged. Try EFI Windows USB Repair. Also this ticket is not connect to this repo. I suggest this to be closed.

closed 2020-04-20 by kecinzer