kecinzer/hpelitebook850g5-opencore #6: It works also on 830 g5

Joaotcs, 2020-04-17:
Hey! With your EFI, it also works on 830G5. But I lost the touchscreen capabilities..
How can I fix these? I think It's a VoodooCHID device.

Joaotcs, 2020-04-17:
Ethernet also doesn't work.
And in my computer, sleep doesn't work.. Videos stop after waking up (for example, after closing chrome, and open again, it doesnt start. videos on vlc, quicktime etc doesnt start. Hackintool crashes...). 
Might be because it's a 8350U (KabyLake Refreshed) with UHD620 (its not HD620).

Joaotcs, 2020-04-17:
IntelMausi (and adding on config) fixed the ethernet.
I still don't have touchscreen (i think I have to edit the SSDT-I2C but don't know how to add the touchscreen device..) and Sleep/wake problem. I have changed framebuffer but still no work.

Joaotcs, 2020-04-17:
Ok, so I have made some changes on framebuffer, managed to get 2048mb on about Mac (used framebuffer of hackintool). And I have also changed to defaults BIOS and only disable security option. Guess what? Wake/Sleep works now! So I suspect it was something on BIOS that was doing this (maybe some hidden features that get "corrupted" after bios update.
Camera is recognized but doesn't show image (clover also did this). Thunderbolt I didn't test but it seems ints not working (no TXHC on hackintool detected). SD card reader etc it doesn't work and will never work.
So I would like only to have my touchscreen working again. Can you help me on this?

Best regards, and great job!

kecinzer, 2020-04-17:
Hi, it's nice to read this.
About touchscreen issues, I think it's related do I2C polling mode. I just tested new version of VoodooI2CHID that works great in interrupt mode so I do changes.

About problem with sleep and videos, it must be something in BIOS because I don't have those issues. You can see on BIOS photos on front page.

Joaotcs, 2020-04-17:
Yes, it was bios problem I have solved it by going default, and redo the changes :)

I also have this problem on start up (just before opencore menu): "ocs: failed to parse string field of type 0". I have seen on google that type 1 and 2 are smbios problems. But I have changed the SMBIOS to clover ones and I still have that error. But it boots up fine.

Also, this is my framebuffer which works with 2048mb nvram, if you want to test it yourself:
<key>PciRoot(0x0)/Pci(0x2,0x0)</key>
			<dict>
				<key>AAPL,GfxYTile</key>
				<data>
				AQAAAA==
				</data>
				<key>AAPL,ig-platform-id</key>
				<data>
				AAAWWQ==
				</data>
				<key>AAPL,slot-name</key>
				<string>Internal@0,2,0</string>
				<key>device-id</key>
				<data>
				FlkAAA==
				</data>
				<key>device_type</key>
				<string>VGA compatible controller</string>
				<key>disable-external-gpu</key>
				<data>
				AQAAAA==
				</data>
				<key>enable-hdmi20</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-con1-enable</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-con1-pipe</key>
				<data>
				CgAAAA==
				</data>
				<key>framebuffer-con1-type</key>
				<data>
				AAgAAA==
				</data>
				<key>framebuffer-fbmem</key>
				<data>
				AACQAA==
				</data>
				<key>framebuffer-patch-enable</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-stolenmem</key>
				<data>
				AAAwAQ==
				</data>
				<key>framebuffer-unifiedmem</key>
				<data>
				AAAAgA==
				</data>
				<key>model</key>
				<string>Intel HD Graphics 620</string>

BTW, in BIOS you only need 64mb to video (in fact 32 would suffice but..).  512mb is unnecessary, it doesn't make anything better and it just steals more RAM, when it's not even needed :)

Joaotcs, 2020-04-17:
And congrats! We can have a Vanilla experience with this!

Joaotcs, 2020-04-17:
Touchscreen and responsiveness of touchpad much better with the recent change you ahve made! Many thanks!

Only "problem" is the  "ocs: failed to parse string field of type 0" error message (twice) before entering OC menu.

I didn't test thunderbolt, but USB-C works. But on hackintool, in USB tab, I only see XHC Sunrise Point-LP usb 3.0, shouldn't there be the thunderbolt drive also?

BTW, things that I changed from yours: Framebuffer (as pointed earlier), changed Misc/Security from 2 to 3 (to be able to use Kext Updater app), added ACPIPooler kext, CPUFriend kext (with idle 800mhz -> works, and saves battery), NotouchID kext, Intelmausi Kext and VoodooTSCsync kext.

It's better than clover now, way cleaner build ;) Thank you.

Joaotcs, 2020-04-18:
[EFI.zip](https://github.com/kecinzer/hpelitebook850g5-opencore/files/4496395/EFI.zip)
Here is my EFI (mod of kecinzer's). Please edit the config with your own serial number! Use clover configurator to generate a valid serial board, system serial and UUID and change the values in config.plist that says CHANGE. (use 15,4 SMBIOS).

No more error in boot (it was an error I had made when I edited the config file.)

Cheers

Rixxx-git, 2020-04-18:
> It's better than clover now, way cleaner build ;) Thank you.

How is it better than clover? How do you recognize it? :)

Joaotcs, 2020-04-18:
> > It's better than clover now, way cleaner build ;) Thank you.
> 
> How is it better than clover? How do you recognize it? :)

For that, you should read about clover vs opencore elsewhere. but I have clearly noticed a better battery management, lower temperatures, faster boot...

Rixxx-git, 2020-04-18:
Alright, thanks! Can I use your EFI also without a Touchscreen on my 830 G5?

Joaotcs, 2020-04-18:
> Alright, thanks! Can I use your EFI also without a Touchscreen on my 830 G5?

Of course, just make sure you backup your current efi to a flashdrive (so you can be able to boot and restore your EFI is something goes wrong), edit my CONFIG.PLIST (change the CHANGE values to your current one if you are already using SMBIOS Macbook pro 15,4, otherwise, open Clover Configurator - DO NOT OPEN THIS CONFIG.PLIST WITH CLOVER CONFIG, just open the app, select smbios  macbook pro 15,4 and randomize serial and copy those values to your new config.plist), delete your efi and replace with this new one, go to bios setup (this is optional, just to make sure you have no bugs because of bios settings) restore default, and then go to bios setup again and just disable security options. When you boot opencore, select MacOS and to make it default boot, click on Ctl+Enter (after this, each boot will go default to MacOS).

Rixxx-git, 2020-04-18:
Where do I find my macaddress?

Rixxx-git, 2020-04-18:
[image]

Rixxx-git, 2020-04-18:
[image]

Joaotcs, 2020-04-18:
Here m8. Enjoy ;) THIS IS ONLY FOR Rixxx-git. Each person must have a unique ID. If someone uses his EFI (this that I am posting) may get a ban from Apple (Rixxx is safe because he will be the first to use and register to his apple account).

Rixxx-git, 2020-04-18:
You can delete the message. I downloaded it. 🙂 Cheers Mate!

Rixxx-git, 2020-04-18:
I will try it now. Do I need the latest BIOS Version? If yes how do I update it?

Rixxx-git, 2020-04-18:
I just exchanged the EFI folder. Without changing my BIOS. All seems to work fine so far. Boot is crazy fast. 😍💪

Joaotcs, 2020-04-18:
Are you sure you have my or kecinzer EFI? It must have ssdt-i2c-hp.aml on acpi folder and Voodoo kexts on Kext folder. Reboot again.. It should be working as you must have the same trackpad as us.
You probably don't need the latest bios, but we do have (installed from windows, or you can create a usb with it, from hp website).

Rixxx-git, 2020-04-18:
Yes I downloaded your EFI folder. I am now updating to the latest Catalina version. Will try to reset the nvram afterwards.

Rixxx-git, 2020-04-18:
Trackpad is working now. :)

Rixxx-git, 2020-04-18:
The webcam is still not working with this setup, right?

Joaotcs, 2020-04-18:
Yes, it is recognized but stays black on my computer. Kecinzer claims in his computer it works. Might be because of different UVC cams used beetween laptops.

Rixxx-git, 2020-04-18:
too bad. But still a huge thank you for your help!!!

Joaotcs, 2020-04-18:
All credits here goes to Kecinzer ;)
Glad I could also help.

Rixxx-git, 2020-04-18:
Do you know if its possible to make the fingerprint scanner work? 🙂

kecinzer, 2020-04-18:
Great to see it works for you.
I just mentioned, that my cpu fan almost not working. I dont know if its BIOS update Q78 V01.10.01 or hackintosh.

Its ok with your configurations?

Joaotcs, 2020-04-18:
Rixx, forget fingerprint, it will NEVER work. There is no fingerprint on any macbooks. (there is no touchscreen also, but it works as a touchpad, so there is no reason why it should  not work. But fingerprint is non exist, so it wont work. Ever.).

Yes, my cpu is doesnt fire up most of the time, but that's just as expected for a real macbook also. I don't play games, so I can't really tell if they are not working for real, or if it's the expected silence for a macbook.

Rixxx-git, 2020-04-19:
> Great to see it works for you.
> I just mentioned, that my cpu fan almost not working. I dont know if its BIOS update Q78 V01.10.01 or hackintosh.
> 
> Its ok with your configurations?

I will see how it goes with my fan. Usually it is working a lot and I didn't update the bios so far. So if the fan works less now it has to be because of OpenCore.

I will let you guys know. 🙂

Rixxx-git, 2020-04-19:
> Rixx, forget fingerprint, it will NEVER work. There is no fingerprint on any macbooks. 

There is a touch ID on Macbook Pro 😉

Rixxx-git, 2020-04-19:
What is the easiest way to update the BIOS? I don't have a Windows installation. (btw what is the Windows folder in your EFI folder there for, @Joaotcs ?)

Joaotcs, 2020-04-19:
> > Rixx, forget fingerprint, it will NEVER work. There is no fingerprint on any macbooks.
> 
> There is a touch ID on Macbook Pro 😉

Too recent that, and is a different piece of hardware. Do not count with this ever working. I mean, we can't even make webcam work. You have to be "lucky" to have a device that works with mac drivers or you must be an expert in reverse engineering to make it work.

My EFI had windows boot folder because I have also windws. You can delete it ;) 

You have to use another p with windows and make a bootable USB installer to use on your laptop. There is no other way, that I am aware of.

kecinzer, 2020-04-20:
Now I want to get work automatic brightness is it working for you? I know that it works in the past...

Joaotcs, 2020-04-20:
> Now I want to get work automatic brightness is it working for you? I know that it works in the past...

Nope, doesn't work. Lighsensor not working.

Rixxx-git, 2020-04-21:
> EDIT2: Mah, I don't see any difference... Anyway, I think they are BIOS controlled, I hope it fires up when temps go high.

How is the temperature going for you? Does the fan work properly?

Mine is still running a lot. So OpenCore doesn't seem to be the solution for this.

Rixxx-git, 2020-04-21:
Is that the right BIOS Update file? 
[image]

If so, how do I create a bootable usb installer with it on my windows laptop?

kecinzer, 2020-04-22:
> Is that the right BIOS Update file?
> [image]
> 
> If so, how do I create a bootable usb installer with it on my windows laptop?

The best, and how I did it, is connect your laptop to UTP cable and make update over internet. This update was very long for me and not working over USB drive.
I have feeling that after this huge update my laptop is much cooler, fan not spinning too much. When I compare to MacBook Pro 15" 2014 its about 70% quieter and laptop body not as hot.

Rixxx-git, 2020-04-22:
> The best, and how I did it, is connect your laptop to UTP cable and make update over internet. This update was very long for me and not working over USB drive.

You mean with a windows partition on mac hackintosh? Start windows, connect ethernet cable and do the BIOS update?

kecinzer, 2020-04-22:
> > The best, and how I did it, is connect your laptop to UTP cable and make update over internet. This update was very long for me and not working over USB drive.
> 
> You mean with a windows partition on my hackintosh? Start windows, connect ethernet cable and do the BIOS update?

No, I mean over you BIOS. Just go to BIOS and proceed from there.

kecinzer, 2020-04-22:
@Joaotcs About 64MB video RAM. I found somewhere, that if you use high resolution you can set as much as you can. As I have 3 displays I need more graphic power :).

Joaotcs, 2020-04-22:
I have played one game of league of legends and it fired once or twice the fans, but not at max power. So I can confirm they are working. Yes, my computer is much quieter, just like a real mac. Idle power (with my CPUFriend kext) is way lower wattage than in windows, and just as smooth.

About 64mb, I have read that more than that doesn't change anything. You graphics will increase Virtual Ram as needed, and using 512mb on BIOS will just perma block 512mb of ram. But, as you have anyway 32gb, there is no problem probably. But I am sure, that if you put 64mb, at least, you won't notice a decrease of performance (but you may also not see any benefit from it).

And if you make the same framebuffer as I, you will get real 2gb of VRAM instead of the 1.5gb.

I have been able to update bios from windows. But I do know that you can use either a bootable USB or by ethernet from bios.

Regards.

Rixxx-git, 2020-04-22:
> No, I mean over you BIOS. Just go to BIOS and proceed from there.

Alrighty, so boot into the BIOS and just choose "update" there? That's it? :)

btw: How do I start the BIOS in OpenCore?

kecinzer, 2020-04-23:
> Alrighty, so boot into the BIOS and just choose "update" there? That's it? :)
>
> btw: How do I start the BIOS in OpenCore?

If you don't know how to get to BIOS, I thing Hackintosh is not good idea for you.
But to get BIOS just reboot laptopt and when you see HP logo hit ESC then select BIOS setup.

kecinzer, 2020-04-23:
> I have played one game of league of legends and it fired once or twice the fans, but not at max power. So I can confirm they are working. Yes, my computer is much quieter, just like a real mac. Idle [...]

You explanation seems legit :). I would like to try your setting. Can you please transform it for opencore config.plist? :).
BTW. why you disabling external GPU? As our laptopot has TB3 port, we can use external GPU for gaming :).

Rixxx-git, 2020-04-23:
> If you don't know how to get to BIOS, I thing Hackintosh is not good idea for you.
> But to get BIOS just reboot laptopt and when you see HP logo hit ESC then select BIOS setup.

Got it. 😉 I am new to Hackintosh and I thought there might be a difference how I get to the BIOS between clover and opencore. But I now understand that has nothing to do with it. 

Update was successfull. Thanks!

Joaotcs, 2020-04-23:
> You explanation seems legit :). I would like to try your setting. Can you please transform it for opencore config.plist? :).
> BTW. why you disabling external GPU? As our laptopot has TB3 port, we can [...]

No real reason xD Initially, it was just to be sure it had no bugs. Also, as I do not have any adapter case to use an External GPU, I forgot to delete that on config.plist (Cuz i am not using it), but it should be working probably (didn't test, but I suppose you did ;p ).
Regards

kecinzer, 2020-04-24:
@Joaotcs what about DRM its working for you? I just tested Apple TV+ and it not working for me.

Joaotcs, 2020-04-25:
I have played trailers. But I don't pay for apple TV.
It seems to be working.

kecinzer, 2020-04-25:
I can play trailers too, but normal content not working.

Joaotcs, 2020-04-25:
I am probably going to change to SMBIOS 14,1 ou 14,3. It should fix it

kecinzer, 2020-04-26:
> I am probably going to change to SMBIOS 14,1 ou 14,3. It should fix it

I dont thing that its related to SMBIOS.

Joaotcs, 2020-05-01:
[EFI.zip](https://github.com/kecinzer/hpelitebook850g5-opencore/files/4564365/EFI.zip)
Here is my EFI (without complete smbios information, please use yours)
I have updated kexts, additional kext and drivers that I find useful (for eg, NTFS capability), I renamed SSDT from 850 to 8X0 (because it is clearly working with all G5 elitebook, at least. And even G6 with minor modification). I have also replaced the open core with N-D-K fork (0.5.7 version also), so we can have two things: A proper GUI (clover like) and, ability to boot Windows or Linux within that GUI!!!
also made one or two modification to config.plist settings (NVRAM and tools mostly).
Cheers.

kecinzer, 2020-05-02:
@Joaotcs thank you for you efi. I will stick with mine. I don’t need GUI for OC. I have only macOS, so I don’t need boot menu at all :). I already disabled guipicker for my configuration.
You can take a look at OpenCanopy boot picker for OpenCore.

Joaotcs, 2020-05-02:
About webcam, I remember that we could use the UV Camera control app to fix webcam, but, as of now, only 64bits app do work on catalina, and that app was only built for 32bits..
And I have not seen anyone capable to build a 64bit of that app as of today.
So we are stuck with the webcam. It's really a hit or miss

Joaotcs, 2020-05-02:
Also, I have been reading about DRM. It seems that on Haswell and newer CPU, if it's iGPU only, it won't work. It seems that Apple is blocking because has specific modifications that Whatevergreen team didn't managed (yet) to bypass/simulate.

jcheunglin, 2020-05-15:
@Joaotcs 
Hi, can you check  the  boot-args, like "-v ". It doesn't work on my 830 G5 with your efi. OpenCore verbose nothing  when i added this argument.
[image]

kecinzer, 2020-05-15:
> @Joaotcs
> Hi, can you check the boot-args, like "-v ". It doesn't work on my 830 G5 with your efi. OpenCore verbose nothing when i added this argument.
> [image]

https://dortania.github.io/OpenCore-Desktop-Guide/troubleshooting/debug.html?h=verbose

Joaotcs, 2020-05-15:
Yeah, -v, is not valid for opencore, use kecinzer code to get verbose working ;) .
Opencore is less "easier" to use, but better when all is stable..

vnmarcel, 2020-05-25:
> Yeah, -v, is not valid for opencore, use kecinzer code to get verbose working ;) .
> Opencore is less "easier" to use, but better when all is stable..

Hi, @Joaotcs ! I managed to boot Catalina 10.15.4 installer using your EFI folder. The installer interface was smooth and I was able to install the system.. But after the installation, when I alredy have the "Macintosh HD", I get stuck at the apple logo... I have the Elitebook G5 Core i5-8350U.. Any tip to make the system work?

[7 more replies]

closed 2020-07-16 by kecinzer