kecinzer/hpelitebook850g5-opencore #10: Works on HP Elitebook 840 G5

igorkolesnikov13, 2020-04-28:
Firstly, thanks a lot for your work.
I was able to get it running with minor modifications for HP Elitebook 840 G5 (8350u/16GB RAM/bcm94352z).

I'm coming from RehabMan's thread, used his method for past 2 years with Mojave and Catalina.

Needed to add Brcm and IntelMausi kexts for Bluetooth and Ethernet.
Everything seems to work nicely (audio input/output, HDMI (audio not tested), WebCam, touch pad with multi-touch, even sip is enabled).

For now, I just have 2 minor issues:
1. Display seems weird. Visually it is fine, no lag or tearing, but it seems too bright or saturated to the point to strain the eyes. It wasn't the case with my previous setup. Does the config.file has some setting for it that I can tweak?
2. Bluetooth seems a bit laggy when using earbuds. Had the same problem previously, just wanted to know if there is some possible solutions.

Once again, thanks for your time.

kecinzer, 2020-04-28:
Great to hear that it works for you.
1. I have no problem at all with it. There is nothing in config.plist, only some patches for HDMI, tak I changed recently and not push changes yet.
2. As I have native Wifi/BT card, I don't have any problem with it (that's why I don't need any kext for it). I using my AirPodsPro daily for Skype for Business meetings.

Joaotcs, 2020-04-29:
Guess you are one of the lucky ones with working webcam ! Bluetooth works fine but I am using bcm94360cs2. You could always purchase this card ;) No kexts needed.

igorkolesnikov13, 2020-04-29:
Actually, it seems, that I found why I have this stutters (mainly with earbuds, mouse is fine). The problem consists that this wireless card (bcm94352z) does not work well when using both Bluetooth and 2.4GHz WIFI.
It seems that using 5GHz WIFI solves the problem.

Unfortunately, at the time when I was buying this card I did not know about bcm94360cs2. If I had, most surely would got it.

vnmarcel, 2020-06-02:
I also got it running on a 840g5 after catalina fresh install.. I ran into some issues with Clover and Opencore after upgrading from Mojave, but had no problemas with a fresh install. I have the bcm94352z wifi card, but ordered a bcm94360ng to try native support

closed 2020-07-16 by kecinzer