kecinzer/hpelitebook850g5-opencore #9: Works on 830 G6

IAmLeo2, 2020-04-28:
Thanks for your work!
This is actually my first Hackintosh and your EFI actually works pretty well on my Elitebook 830 G6 ( similar hardware ) but I have some few quirks like:

- No audio
- Minor graphics issues (Screen tearing in status bars + 7Mb graphics memory)
- Mouse Click not working
- No Bluetooth ( card in supposed to be natively supported )

**My Config:**

HP Elitebook 830 G6

PROCESSOR: Intel Core i5 8365U
SSD : 512GB KIGSTON A2000 (Windows and Linux) + 256GB  NVME KINGSPEC NE-256
RAM : 2 x 8GB Crucial CT2K8G4SFS832A
GPU : iGPU Intel UHD 630
Wifi Card : M.2 Broadcom BCM94360NG

Thanks

kecinzer, 2020-04-28:
I'm afraid, that you need special SSDT files that match your configuration.
Try to start from scratch with OC guide - https://dortania.github.io/oc-laptop-guide/ and more detailed https://dortania.github.io/OpenCore-Desktop-Guide/.

Your GPU is natively supported I thing, so you don't need SSDT-IGPU.aml (but I only suppose).

IAmLeo2, 2020-04-28:
Thanks for your reply.

I am working on my custom SSDT files.

Joaotcs, 2020-04-29:
Mouse Click not working - physical buttons? That and the middle joystick won't work.
You may try to swap your card with BCM94360cs2. For audio, you may have to try to use another apple alc id layout!

kecinzer, 2020-04-30:
@IAmLeo2 take a look at this - https://github.com/acidanthera/AppleALC/pull/496

Joaotcs, 2020-07-03:
Hello @IAmLeo2, can you share your EFI with me? I have acquired another G6, but I am having blackscreen after apple logo on install menu 😔

Joaotcs, 2020-07-04:
Ok, changed PNLF, framebuffer, apple alc layout. Now, I only have the same problem with the bluetooth. It recognizes wifi, but not the bluetooth (in hackintool it doesn't recognize it either). Any idea about this @kecinzer (as I am using a native supported card, that works on g5)?

P.S According to guides, SetupVirtualMap should be active in config.plist (you don't have this @kecinzer). And... do you have any idea if we need HPET fixes ? (from SSDTTime script)

Joaotcs, 2020-07-04:
OOFTOPIC: Elitebook g5 and g6 is almost the same. Even the case is the same except the top cover. Also, motherboard is almost exactly the same. No real updates from HP nor Intel... I guess the move to ARM is a smart move (that will probably kill hackintosh :( ), Intel has not made any real upgrades since 6th gen CPU.

closed 2020-07-16 by kecinzer