kecinzer/hpelitebook850g5-opencore #12: Change of ID device iGPU?

Joaotcs, 2020-06-03:
Hello, I have been reading the new info of Opencore 0.5.9 and it seems that:
"For UHD620 users, you must use:
device-id=87C00000
AAPL,ig-platform-id=0000C087
"

Also, "HP Users:

Kernel -> Quirks -> LapicKernelPanic -> True
You will get a kernel panic on LAPIC otherwise
UEFI -> Quirks -> UnblockFsConnect -> True"

Should we change something? I know that you have used even in acpi files device id: 16590000... Don't know if it makes anything better.
Regards

kecinzer, 2020-06-05:
I start to look at it. Preparing huge clean up. Thanks for pushing me :).

I don’t have touch screen and my touchpad works great.

Joaotcs, 2020-06-05:
Thanks mate, wish I had your knowledge to fix things myself xD. I am learning by seeing ur modifications ;) . Also, can you please tell me how to ad the mod to aml files to be able to boot windows? I have read that I must add something like : "If (_OSI ("Darwin")) ...." so that If I try to boot something that is not macOS, opencore won't try to patch things.

Also, with the updated kext, I have not got again the problem ;)

kecinzer, 2020-06-05:
I’m finishing massive cleanups and whole ssdts would change according to OpenCore guide. Almost nothing left from RehabMan.
Currently only 2 things left.
1) Trackpad GPI0 mode instead of _OSI to XOSI. This is what I want to do, but with SSDT trackpad not working and I don’t know why.
2) RTC post fix acpi reneame convert to rtc-blacklist range. This is very minor and may be I left it as it.

Joaotcs, 2020-06-05:
Man, if you find any guide or knowledge book about this, please post here. I would really like to understand more on this. I can understand config files, but SSDT is way too much for me.
Would like to help, but I am not capable.

Thank you for your hard work and making it public. Rehab was really the man on clover times, made probook and elite,book kings of hackingtosh but he is AFK since 2019. You might become the next hackintosh pro  ;)
Regards

Joaotcs, 2020-06-07:
Hello m8, I also confirm that sleep works, (have done 3 sleeps in a row) with native Thunderbolt. I am still using ur latest opencore (with some mods on config), and bios is basically: Default, no security settings, no fastboot, no ethernet boot, 64mb in video...

kecinzer, 2020-06-08:
> Hello m8, I also confirm that sleep works, (have done 3 sleeps in a row) with native Thunderbolt. I am still using ur latest opencore (with some mods on config), and bios is basically: Default, no security settings, no fastboot, no ethernet boot, 64mb in video...

It's fixed in this commit, do you use it? https://github.com/kecinzer/hpelitebook850g5-opencore/commit/d4d6ac06b93528471737623dd0ae328bdbb9b348

Joaotcs, 2020-06-08:
> > Hello m8, I also confirm that sleep works, (have done 3 sleeps in a row) with native Thunderbolt. I am still using ur latest opencore (with some mods on config), and bios is basically: Default, no security settings, no fastboot, no ethernet boot, 64mb in video...
> 
> It's fixed in this commit, do you use it? [d4d6ac0](https://github.com/kecinzer/hpelitebook850g5-opencore/commit/d4d6ac06b93528471737623dd0ae328bdbb9b348)

Ups, yes, I am. I hadn't read the changelog, I though you were still using the Legacy mode.

Joaotcs, 2020-06-08:
Thanks m8! I will study your guides and review your modifications so I can also learn a bit more.

Much appreciation for your work.

kecinzer, 2020-06-09:
> Thanks m8! I will study your guides and review your modifications so I can also learn a bit more.
> 
> Much appreciation for your work.

Thank you. I'm still learning.
Now I'm trying to make our touchpad to work other way - https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad.html see here, but it not working. I already asked on few forums and send bugreport, but still have no answer.

raulgz04, 2020-06-15:
Thanks for your work @kecinzer , it works almost perfect with the latest EFI, but I have one error, when I plug a usb-c hard disk, it connects like usb 2.0, to HS05 port, and not to USB 3.1, like before this changes for TB3.
I think it could be related with the port number.
I have in native mode already.

raulgz04, 2020-06-16:
I fixed it changing ADR of SSP1 to 0x03, changing HS01 to HS05 and his ADR to 0x05 and deleting SSP2 and HS02

kecinzer, 2020-06-16:
I have changed SSDT for TB3, but I not commited it yet. Currently I’m on vacation. I will later look at your changes. Thanks.

kecinzer, 2020-06-16:
> I fixed it changing ADR of SSP1 to 0x03, changing HS01 to HS05 and his ADR to 0x05 and deleting SSP2 and HS02

Can you please make Pull Request? I would like to merge it :). In fact, I didn’t tested where USB-C drive is connected, only if it works :). Thanks.

raulgz04, 2020-06-16:
I'm sorry I'm new at using GitHub, I don't now how to do that. I would attach the modified SSDT file.
I changed the device-id of the GPU too, as they recommend here:
https://dortania.github.io/vanilla-laptop-guide/OpenCore/config-laptop.plist/kaby-lake.html#deviceproperties
"For UHD620 users we recommend the following values:
device-id=16590000
AAPL,ig-platform-id=0000C087"

[SSDT-TB3.aml.zip](https://github.com/kecinzer/hpelitebook850g5-opencore/files/4789528/SSDT-TB3.aml.zip)

Joaotcs, 2020-06-17:
> I'm sorry I'm new at using GitHub, I don't now how to do that. I would attach the modified SSDT file.
> I changed the device-id of the GPU too, as they recommend here: [...]

You can edit the TB3.dsl file at the SSDT folder of this github, and then pull request ;)

Thanks for sharing it

Joaotcs, 2020-06-17:
About device-id, they must have edited, as when I have read, I device-id should be 87C00000, but they changed it again xD

kecinzer, 2020-06-22:
https://github.com/dortania/vanilla-laptop-guide/commit/9d767e8e7553d553670d02835961b2845f422ff5

closed 2020-07-16 by kecinzer