kecinzer/hpelitebook850g5-opencore #16: Touchpad randomly stops working

Joaotcs, 2020-07-01:
Sometimes the touchpad doesnt work after booting, or after waking up.
Can't really understand why this happens only sometimes.

Joaotcs, 2020-07-01:
Rebooting fixes the problem.Keyboard always works

piKolin, 2020-07-01:
Mine doesn't work inside Open Canopy. For select the boot option i always need to use the Keyboard.

kecinzer, 2020-07-01:
Do you have multiple OS?
This can be related to Windows 10.

I'm still working on different method for touchpad.

piKolin, 2020-07-01:
> Do you have multiple OS?
> This can be related to Windows 10.
> 
> I'm still working on different method for touchpad.

Yes, i need W10 for work reasons. i haven't tested Open Canopy only with Catalina installed

Joaotcs, 2020-07-01:
> Mine doesn't work inside Open Canopy. For select the boot option i always need to use the Keyboard.

Thats just cuz you dont have PS2 drivers on your EFI file. But the problem is in the OS itself.

> Do you have multiple OS?
> This can be related to Windows 10.
> 
> I'm still working on different method for touchpad.

Yes, I have. Ok, so that's probably the cause ok. I have also tried to changed the method for touchpad, but I couldn't make it work, dunno why...

kecinzer, 2020-07-02:
Ok, try it with this commit - https://github.com/kecinzer/hpelitebook850g5-opencore/commit/30a6c8c8cd18db4edcc9ab658481d4e014a1f5d6

@piKolin sorry, but my configuration not use OpenCanopy neither I, so this issue is not relevant here. Try OpenCore support.

Joaotcs, 2020-07-02:
> Ok, try it with this commit - [30a6c8c](https://github.com/kecinzer/hpelitebook850g5-opencore/commit/30a6c8c8cd18db4edcc9ab658481d4e014a1f5d6)
> 
> @piKolin sorry, but my configuration not use OpenCanopy neither I, so this issue is not relevant here. Try OpenCore support.

Will have to test it more, but It seems that touchscreen stops working. It's also an I2C device. Touchpad is working atm, but need more time, as it was also working before, it was on some boots or some awakes that the problem appeared.

Joaotcs, 2020-07-02:
OK, it happened again.. After waking up, touchpad doesn't work.. And touchscreen never works. Reverting to the previous version.

kecinzer, 2020-07-02:
> OK, it happened again.. After waking up, touchpad doesn't work.. And touchscreen never works. Reverting to the previous version.

This new method is related to only touchpad device as I dont have touch screen. For your concrete device you need to check your DSDT and find places that need to patch.
But touchpad still have this issue I think first method is better for you.

Joaotcs, 2020-07-03:
> This new method is related to only touchpad device as I dont have touch screen. For your concrete device you need to check your DSDT and find places that need to patch.
> But touchpad still have this [...]

What should I patch?
The touchscreen device is the TLP0.
Here is my DSDT section - can you help me with the rename patch?
   Scope (_SB.PCI0.I2C0)
    {
        Device (TPL0)
        {
            Name (HID2, One)
            Name (SBFB, ResourceTemplate ()
            {
                I2cSerialBusV2 (0x0010, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                    0x00, ResourceConsumer, _Y2A, Exclusive,
                    )
            })
            Name (SBFG, ResourceTemplate ()
            {
                GpioInt (Level, ActiveLow, Exclusive, PullDefault, 0x0000,
                    "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                    )
                    {   // Pin list
                        0x0000
                    }
            })
            Name (SBFI, ResourceTemplate ()
            {
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, _Y2B)
                {
                    0x00000000,
                }
            })
            CreateWordField (SBFB, \_SB.PCI0.I2C0.TPL0._Y2A._ADR, BADR)  // _ADR: Address
            CreateDWordField (SBFB, \_SB.PCI0.I2C0.TPL0._Y2A._SPE, SPED)  // _SPE: Speed
            CreateDWordField (SBFI, \_SB.PCI0.I2C0.TPL0._Y2B._INT, INT2)  // _INT: Interrupts
            CreateWordField (SBFG, 0x17, INT1)
            Method (_INI, 0, NotSerialized)  // _INI: Initialize
            {
                If (LLess (OSYS, 0x07DC))
                {
                    SRXO (GPLI, One)
                }

                Store (GNUM (GPLI), INT1)
                Store (INUM (GPLI), INT2)
                If (LEqual (SDM0, Zero))
                {
                    SHPO (GPLI, One)
                }

                Store (THHA, HID2)
                Store (TPSA, BADR)
                Store (0x00061A80, SPED)
            }

            Name (_HID, "ELAN25D0")  // _HID: Hardware ID
            Name (_CID, "PNP0C50")  // _CID: Compatible ID
            Name (_S0W, 0x04)  // _S0W: S0 Device Wake State
            Method (_DSM, 4, Serialized)  // _DSM: Device-Specific Method
            {
                If (LEqual (Arg0, HIDG))
                {
                    Return (HIDD (Arg0, Arg1, Arg2, Arg3, HID2))
                }

                If (LEqual (Arg0, TP7G))
                {
                    Return (TP7D (Arg0, Arg1, Arg2, Arg3, SBFB, SBFG))
                }

                Return (Buffer (One)
                {
                     0x00                                           
                })
            }

            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                Return (TPLE)
            }

            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                If (LLess (OSYS, 0x07DC))
                {
                    Return (SBFI)
                }

                If (LEqual (SDM0, Zero))
                {
                    Return (ConcatenateResTemplate (SBFB, SBFG))
                }

                Return (ConcatenateResTemplate (SBFB, SBFI))
            }
        }
    }

kecinzer, 2020-07-03:
You need to patch same methods as for CPD0. For this, you need to find hexa codes in DSDT.aml. But in this situation, I think previously used method (_OSI to XOSI) rename is much better and universal for more devices. I will revert to it.
This renames do almost same thing but need more work. It definetely dont solve your problem with randomly stop working trackpad.
Do you place question on VoodooI2C support?

closed 2020-07-16 by kecinzer

borygo77, 2020-08-19:
It's related to new voodooI2C release. After updating its same on mine. Downgraded and it works with no problem ;)