kecinzer/hpelitebook850g5-opencore #17: Bad battery life

kecinzer, 2020-07-09:
I mentioned few days earlier, that I get very bad battery life. I don't know since when this start happening, but I have suspicion about "Massive overhaul".
What is yours battery life?

Se7Touch, 2020-07-10:
i have elitebook 840 g5 i7 32gb and perfect battery life with this EFI :/

(avg. 6h-8h)

Rixxx-git, 2020-07-17:
that is a good question. I am on a 830 g5 and my batterie only holds for about 3 hours.

Joaotcs, 2020-07-18:
around 6 hours. 830 g5. I have CPUFriend

Rixxx-git, 2020-07-19:
> around 6 hours. 830 g5. I have CPUFriend

What is CPUFriend and how can I add it to my setup? :)

kecinzer, 2020-07-19:
I already have CPUFriend in my latest version.

Rixxx-git, 2020-07-20:
Oh, right. 😄 But I still don't get 6 hours battery life. Do you have anything different @Joaotcs?

levsha-me, 2020-07-23:
> i have elitebook 840 g5 i7 32gb and perfect battery life with this EFI :/
> 
> (avg. 6h-8h)

Hey. I have an identical laptop - EliteBook 840 G5 i7 (32Gb RAM) and my battery lasts only about 2-3 hours maximum. But could you share your EFI?

Rixxx-git, 2020-07-23:
Strange. How can that be? Different battery life with the same laptop and EFI setup?

levsha-me, 2020-07-23:
> Strange. How can that be? Different battery life with the same laptop and EFI setup?

I use OpenCore, but since a person on Clover has much better autonomy, because of this I ask you to share your EFI folder

Rixxx-git, 2020-07-23:
Ah ok. I thought they are all using OpenCore.

Rixxx-git, 2020-07-27:
unfortunately no help yet. :/ Do you use clover or opencore @Joaotcs ?

Joaotcs, 2020-07-27:
Opencore, latest bios version, with balance power in bios..

Rixxx-git, 2020-07-27:
Thank you! I have the latest bios as well. 🙂  What do you mean with balance power?

Joaotcs, 2020-07-31:
Maximize my battery duration... And all the idle stuff turned on..

kecinzer, 2020-11-12:
I done some more tweaks but I still get around 5 hours of battery, but it should be around 9 hours. I found setting in BIOS that Turbo boost is turned off on battery, but it was already enabled.
I don’t know what to do next.

Rixxx-git, 2020-11-13:
same for me

kecinzer, 2020-11-18:
With SMBus SSDT I can reach 6 hours from real test :). But I think it’s still not enough.

Joaotcs, 2020-11-23:
That's probably the best we can get... I have seen similar problems before. There is no solution beside what you have already done. You have optimized to the best possible, to my knowledge... Can't recall anything to optimize - maybe disabling Thunderbolt, but you need them. Battery life will always be better on Windows side..

kecinzer, 2020-12-04:
Ok, I let my college with exactly same model but with Windows 10 do the battery life test. He can't even reach 6 hours of battery life. So I think it's ok.

closed 2020-12-04 by kecinzer