kecinzer/hpelitebook850g5-opencore #18: New model and some problems.

Se7Touch, 2020-07-10:
Hi bro,

Im use Catalina with your EFI on HP EliteBook 840 G5, its works perfectly but i found 2 issue.

1. I cant use Ethernet its not work with IntelMausi kext, I tried different versions, but I did not get any results.
2. When I try to select text with the touchpad, the cursor moves around randomly.

Thank you for hard work, good luck!

kecinzer, 2020-07-16:
Hi!

1. Make sure you have Intel ethernet card.

1.

Se7Touch, 2020-07-16:
Thank you for answer to me!

1. Yes, my ethernet card is Intel I219-LM but not work with any kext.
2. I changed touchpad settings and now works perfectly. TY!!

[image]

kecinzer, 2020-07-16:
Great that you fixed touchpad. I completly forgot about force click settings.

With LAN its weird. May be you forgot something in BIOS (LAN/WLAN switching needs to be off)?

Se7Touch, 2020-07-17:
Pfff. Its my bad.
When I checked my BIOS settings, I noticed that the embedded LAN controller is off. I have no idea why I closed it.
I no longer have to use a USB LAN adapter.

Thanx for all... I will continue to be your follower for developments.

kecinzer, 2020-07-17:
Great. I'm happy that everything works for you now :).

closed 2020-07-17 by kecinzer