kecinzer/hpelitebook850g5-opencore #19: HDMI/USB-C hub connection problems

igorkolesnikov13, 2020-07-15:
Hi,

I am still following your awesome job! Thanks a lot.

I just wanted to ask if anyone is facing the same issue.
After the last update (15.05/commit#43), I have a very strange behavior with my HDMI display.

If I connect it via UBS-C hub with HDMI port (Ugreeen), it will become disconnected right after the display turns off, or the computer goes to sleep. The only solution is to reconnect the cable.
If I connect it via internal HDMI port, it wont be recognized on reboot. The only solution is to reconnect the cable.

Is any one facing it?

kecinzer, 2020-07-15:
Hi @igorkolesnikov13 can you please try add argument igfxonln=1 to boot-args in NVRAM section?

<key>boot-args</key>
<string>debug=0x100 igfxonln=1</string>

And try it with that.

igorkolesnikov13, 2020-07-15:
Hey,

Thanks for a prompt reply.
I tried some thing already:
1. I added the suggested arguments to boot-args
debug=0x100 keepsyms=1 igfxonln=1 -disablegfxfirmware
It does work for the internal HDMI port, but not for USB-C

2. I tried to changing CNConnectorAlwaysConnected in Hackintool -> Patch -> Connectors
But for some reason, it doesn't seem to apply the settings.
Here are some screens: 
https://imgur.com/fHXkhEp 
https://imgur.com/3SwyQ6C

Before the update, I was using 15.04 and quite an old commit. One before massive overhaul. I think it was "End of April Update."

kecinzer, 2020-07-15:
> Hey,
>
> Thanks for a prompt reply.
> I tried some thing already:
>
> 1. I added the suggested arguments to boot-args
>    debug=0x100 keepsyms=1 igfxonln=1 -disablegfxfirmware
>    It does work for the [...]

OK, next think you can try is disable these 2 SSDTs:

<dict>
				<key>Comment</key>
				<string>DTGP helper method</string>
				<key>Enabled</key>
				<true/>
				<key>Path</key>
				<string>SSDT-TB3-DTGP.aml</string>
			</dict>
			<dict>
				<key>Comment</key>
				<string>Thunderbolt 3 driver assignment</string>
				<key>Enabled</key>
				<true/>
				<key>Path</key>
				<string>SSDT-TB3.aml</string>
			</dict>

Then let me know :).

kecinzer, 2020-07-16:
I do little research about this and it seems that it must be something in macOS update. I tried disable TB3 SSDTs, even change TB mode in BIOS, but USB-C to HDMI output not work after display sleep.
I’m pretty sure, that it worked in the past.

igorkolesnikov13, 2020-07-16:
I tried to disable the two SSDTs that you pointed to. The problem persisted. Actually, it got a bit strange. Sometimes it worked and sometimes no.

As for the update, yes, up until 15.02 it was working perfectly, then, I updated directly to 15.04 and needed to add debug=0x100 igfxonln=1 to boot-args. And now this.

kecinzer, 2020-07-16:
I play with it, but still can't figure out what it causing. Luckily for me, I use TB3 -> Dupal DP dongle that works great.
So for now I will hope, that someone found fix for this.

Joaotcs, 2020-07-18:
This is probably a whatevergreen fault no? Might have to do some rework with connector and busid...

Joaotcs, 2020-07-27:
10.15.4 update broke a lot of hdmi connections.. Display port seems to work anyway

Joaotcs, 2020-08-01:
This is my framebuffer. See if it helps...
				<key>AAPL,ig-platform-id</key>
				<data>
				AAAWWQ==
				</data>
				<key>AAPL,slot-name</key>
				<string>Internal@0,2,0</string>
				<key>device-id</key>
				<data>
				ElkAAA==
				</data>
				<key>device_type</key>
				<string>VGA compatible controller</string>
				<key>disable-external-gpu</key>
				<data>
				AQAAAA==
				</data>
				<key>force-online</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-con1-enable</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-con1-type</key>
				<data>
				AAgAAA==
				</data>
				<key>framebuffer-patch-enable</key>
				<data>
				AQAAAA==
				</data>
				<key>framebuffer-unifiedmem</key>
				<data>
				AAAAgA==
				</data>
				<key>hda-gfx</key>
				<string>onboard-1</string>
				<key>model</key>
				<string>Intel UHD Graphics 620</string>

igorkolesnikov13, 2020-08-01:
Yeah, after 10.15.4 it got really bad. I'm waiting for the next release of kexts, if it won't be fixed, I will probably roll back to 10.15.3.

Thanks for the framebuffer. Can you point me to the guide on how to test it?

Joaotcs, 2020-08-01:
Open config.plist, and edit in Deviceproperties the PCIroot corresponding to graphics (just delete the code you have inside the graphics with this one ;)
Please backup your EFI on a usb drive in case you mess up

kecinzer, 2020-08-24:
@Joaotcs OMG, you are my lifesaver!
Because I don't want to mess with cables and plugging/replugging it I bought this dock station - https://i-tec.cz/en/produkt/tb3cdualdppdx-2/ because I never had simple problem with this adatper - https://i-tec.cz/en/produkt/tb3dual4kdp-2/.

But I was really angry, when I set up this new dock station and figure out, that only one external display wake up after sleep. So I start to play with frame buffers. I tried several for our UHD 620 and recommanded values and I tried your config for fun. Now everything working like a charm! Thank you a lot!

I don't know how it's possible, because your values of device-id and AAPL,ig-platform-id is different than those in tutorials and FAQ. Where you get those values? But never-mind, don't touch running system!

Thank you again. I'll do some more tests (especially with my USBC- HDMI cable) and merge changes.

Joaotcs, 2020-08-24:
Glad I could help ;)
I can't really remember, but it was days of testing and reading a lot of different threads, I came a cross with many different fixes, tested them all, made some modifications and I ended up with this. I have also been successful with elitebook g6, but sold the pc cuz of audio and keyboard hot keys incompatibility.

kecinzer, 2020-08-26:
@igorkolesnikov13 
Please try USB-C - HDMI with latest commit - https://github.com/kecinzer/hpelitebook850g5-opencore/commit/4d9fe93fb0bc5061d8c7e4937f075bed4166274a

igorkolesnikov13, 2020-08-26:
Hi, thanks for the tip.

I got a lot of other errors on 10.15.6 and a bit short of time to debug.
For now, I'm on 10.15.3 with your latest commit. This version works best for now.
Maybe I'll wait for Big Sur to see how it works.

kecinzer, 2020-08-27:
It really looks like macOS issue - https://dortania.github.io/OpenCore-Post-Install/universal/sleep.html#fixing-gpus

Joaotcs, 2020-08-27:
What problems do you have atm @kecinzer ? The only thing that I have is sometimes, after wake, touchpad doesnt work. Someone said it's because of latest I2C kexts. Might try someday older releases. I am thinking that updating kext, at least keyboard/touchpad wise is not worth the hassle when it is already working perfectly..

kecinzer, 2020-08-28:
@Joaotcs I have only major problem with graphics. My USB-C to HDMI now not work at all. My docking station works normally when I lock my macOS (put monitors to sleep). But when I put laptop into sleep and wakes it, 2 external displays act as 1 with mirroring. And integrated LAN card in dock station not working properly - it has self assigned IP address.
I need to turn of and on dock station to start working again.

kecinzer, 2020-12-04:
As I switched completly to TB3 hub and adapter and there are many changes in Bug Sur, I'm closing this issue.

closed 2020-12-04 by kecinzer