kecinzer/hpelitebook850g5-opencore #20: Using your EFI Folder

Rixxx-git, 2020-07-17:
Hi buddy,

I just replaced my EFI and OC folder with yours and now I get this on startup. 

What did I do wrong? 

Everything was running fine before. Just updated to the latest MacOS. 

Thank you for your help! 

[image]

Rixxx-git, 2020-07-17:
[image]

kecinzer, 2020-07-17:
You need to place only content of EFI folder from this repo - OC folder and BOOT folder.
SSDT folder are only decompiled ACPI files.

Rixxx-git, 2020-07-17:
Ah alright. My bad. Thanks for the quick reply!

I replaced your config.plist with mine. Isn't that right?

kecinzer, 2020-07-17:
> Ah alright. My bad. Thanks for the quick reply!
> 
> I replaced your config.plist with mine. Isn't that right?

No, you need to adopt your configuration, as I changed many files - kexts and also ACPI files.
Your config.plist may pointing to files, that no more exists in current version.

Rixxx-git, 2020-07-17:
Alright. Thank you!

Which lines of my config file do I have to adopt? :)

Rixxx-git, 2020-07-17:
I am trying to start from my super-duper backup on an external hard drive but I get the same start menu as before. 

Why is that? 

I am trying to start my former MacOS setup to fix the problem like you said. 

[image]

Rixxx-git, 2020-07-17:
What do I have to do to be able to fix the problem? :(

Rixxx-git, 2020-07-17:
got it working again. which lines do I have to change in your config-plist file with my 830 g5? 🙂

Rixxx-git, 2020-07-17:
all good now. thanks!

closed 2020-07-17 by Rixxx-git