kecinzer/hpelitebook850g5-opencore #22: Partly successful with 840g6

everyb313, 2020-08-06:
@kecinzer: Thanks for your valuable work ! Even a beginner like me was able to modify !

**Changes for 840G6 (CPU i5 8265U, Catalina 10.15.6)**
------------------------------------------------------------------------------------------
**Display** 
device section:
ig-platform-id to 0900A53E
device-id to        A53E0000
ACPI section:
add SSDT-PNLFCF (source github: baohavan/hpelitebook840g6-opencore, otherwise the display brightness is very low)

drawback: cannot change display brightness by F3/F4

**Audio**
device section:
change layout-id to 18

drawback: no microphone, Mic button F8 opens apple music application

**No camera** 
(no idea why not)
------------------------------------------------------------------------------------------

Any hints ?
BR everyb313

borygo77, 2020-08-17:
I got x360 830G6 and same like you mic h=and camera doesn't work. Removed ssdt-input as those are only remaps done by kecinzer. Brightness keys should work out of the box even with ssd-input. Sometime need to reset EC to get back brightness keys back, just hold power button for 30 sec ;)

frax256, 2020-11-02:
I can't get it it to run at all on my 840 G6, just gives me a black screen and boots to the next available OS installed.
Any ideas?

kecinzer, 2020-11-03:
Try to enable bootpicker. As I use only macOS without any other OS, I don't need it.

closed 2020-12-04 by kecinzer