kecinzer/hpelitebook850g5-opencore #23: Apps hanging after sleep

kecinzer, 2020-08-10:
After latest OC 0.6.0 update, I mentioned, that I can't sleep. After I wake up from sleep, many application start to not responding and I need to force close it and it can't be opened again, only after reboot.

I tried to restore to OC 0.5.9 with its kexts, but is still same. I don't understand what is wrong.

kecinzer, 2020-08-10:
Fixed by this - https://github.com/acidanthera/bugtracker/issues/990#issuecomment-671273436

Rixxx-git, 2020-08-14:
just fyi: There is no such option in the BIOS of my Elitebook 830 G5. At least I can't find it.

Joaotcs, 2020-08-19:
> Fixed by this - [acidanthera/bugtracker#990 (comment)](https://github.com/acidanthera/bugtracker/issues/990#issuecomment-671273436)

Yes, that is the option of bios that gave all those after sleep problems. Restoring BIOS turned on that fuction. I already had that conclusion before, though it was already known :s

Joaotcs, 2020-08-19:
> just fyi: There is no such option in the BIOS of my Elitebook 830 G5. At least I can't find it.

I guarantee you, you do have. I think it's one of the last option of bios settings. Just search for it.

closed 2020-09-01 by kecinzer

TCB13, 2022-06-10:
> > just fyi: There is no such option in the BIOS of my Elitebook 830 G5. At least I can't find it.
> 
> I guarantee you, you do have. I think it's one of the last option of bios settings. Just search for it.

Hmm.. or maybe not.

[image]

When I turn on AMT I get Intel ME options in the first screen, so I guess HP decided to bundle it all together. Either way it doesn't fix the sleep issues in my case.