kecinzer/hpelitebook850g5-opencore #26: High temperature under stress and trottling.

artenidas, 2020-08-28:
Hello guys. Have question what temperatures u have on stress. I using Liquid metal on CPU and have 80-85st celcius after run cinebech. As soon as the processor reaches 80st, it goes into trotting and breaks the clock. Can't undervolt CPu with voltageshift becouse program say OC_Locked can only decrease power watt package(But it doesn't give too much) If i working on Windows i undervolt this procesor and dont have problem with trottling. Any Solution? Maybe u have too this problem?

kecinzer, 2020-12-06:
I need to try this, but I never had problem with throttling.
But I looked at voltageshif and I wanted to undervolt because of battery life, but I'm afraid, that we have this option locked by BIOS (OC_Locked).
I don't understand how you can undervolt this CPU in Windows, because its BIOS related.

kecinzer, 2020-12-07:
Ok, I just tried Cinebench. I can see temperatures around 70°C, may be system throttling, but I think it's ok.
My results is CPU Multi: 2636 pts and single 861 pts. MP Ratio 3,06.
I can normally work with Cinebench, so I'm happy with this result.

kecinzer, 2020-12-10:
Tested on Windows 10 and I got little less score that on macOS:
CPU Multi: 2588 pts
CPU single: 880 pts
MB Ration: 2.94