kecinzer/hpelitebook850g5-opencore #27: How to add touchscreen

Joaotcs, 2020-10-02:
Hi! I wanted to try your EFI, but it seems that I can't make it work now... How to add Touchscreen? I tried to make one for my touchscreen according to your new .aml files, but wasn't capable :/
My device: 
IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/I2C0@15/VoodooI2CPCIController/VoodooI2CControllerNub/VoodooI2CControllerDriver/TPL0

[image]

Joaotcs, 2020-10-02:
fixed! Close topic

closed 2020-10-02 by Joaotcs

kecinzer, 2020-10-05:
Wow, that was fast. Can you please share some results? If anyone else need touchscreen?

Joaotcs, 2020-10-05:
Sure ;) I have tested on another Touchscreen and it worked!

Joaotcs, 2020-10-05:
So you need to add this TPL0.aml to config.plist:
			<dict>
				<key>Comment</key>
				<string>I2C Touchscreen</string>
				<key>Enabled</key>
				<true/>
				<key>Path</key>
				<string>SSDT-TPL0.aml</string>
			</dict>
and replace rename
			<dict>
				<key>Comment</key>
				<string>Change _INI to XINI of CPD0 and TPL0 device</string>
				<key>Count</key>
				<integer>2</integer>
				<key>Enabled</key>
				<true/>
				<key>Find</key>
				<data>
				X0lOSQCgEpVPU1lT
				</data>
				<key>Limit</key>
				<integer>0</integer>
				<key>Mask</key>
				<data>
				</data>
				<key>OemTableId</key>
				<data>
				</data>
				<key>Replace</key>
				<data>
				WElOSQCgEpVPU1lT
				</data>
				<key>ReplaceMask</key>
				<data>
				</data>
				<key>Skip</key>
				<integer>2</integer>
				<key>TableLength</key>
				<integer>0</integer>
				<key>TableSignature</key>
				<data>
				</data>
			</dict>
			<dict>
				<key>Comment</key>
				<string>Change _CRS to XCRS of CPD0 and TPL0 device</string>
				<key>Count</key>
				<integer>2</integer>
				<key>Enabled</key>
				<true/>
				<key>Find</key>
				<data>
				X0NSUwCgDpVPU1lT
				</data>
				<key>Limit</key>
				<integer>0</integer>
				<key>Mask</key>
				<data>
				</data>
				<key>OemTableId</key>
				<data>
				</data>
				<key>Replace</key>
				<data>
				WENSUwCgDpVPU1lT
				</data>
				<key>ReplaceMask</key>
				<data>
				</data>
				<key>Skip</key>
				<integer>2</integer>
				<key>TableLength</key>
				<integer>0</integer>
				<key>TableSignature</key>
				<data>
				</data>
			</dict>

Joaotcs, 2020-10-05:
Here is TLP0.aml:

/*
 * Intel ACPI Component Architecture
 * AML/ASL+ Disassembler version 20180427 (64-bit version)(RM)
 * Copyright (c) 2000 - 2018 Intel Corporation
 * 
 * Disassembling to non-symbolic legacy ASL operators
 *
 * Disassembly of iASLPcBdb7.aml, Mon Oct  5 12:28:20 2020
 *
 * Original Table Header:
 *     Signature        "SSDT"
 *     Length           0x00000262 (610)
 *     Revision         0x02
 *     Checksum         0xD7
 *     OEM ID           "ACDT"
 *     OEM Table ID     "TPL0"
 *     OEM Revision     0x00000000 (0)
 *     Compiler ID      "INTL"
 *     Compiler Version 0x20180427 (538444839)
 */
DefinitionBlock ("", "SSDT", 2, "ACDT", "TPL0", 0x00000000)
{
    External (_SB_.GNUM, MethodObj)    // 1 Arguments (from opcode)
    External (_SB_.INUM, MethodObj)    // 1 Arguments (from opcode)
    External (_SB_.PCI0.I2C0.TPL0, DeviceObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.BADR, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.HID2, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.INT1, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.INT2, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.SBFB, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.SBFG, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.SBFI, IntObj)    // (from opcode)
    External (_SB_.PCI0.I2C0.TPL0.SPED, IntObj)    // (from opcode)
    External (_SB_.SHPO, MethodObj)    // 2 Arguments (from opcode)
    External (_SB_.SRXO, MethodObj)    // 2 Arguments (from opcode)
    External (GPLI, FieldUnitObj)    // (from opcode)
    External (OSYS, FieldUnitObj)    // (from opcode)
    External (SBFI, IntObj)    // (from opcode)
    External (SDM0, FieldUnitObj)    // (from opcode)
    External (THHA, FieldUnitObj)    // (from opcode)
    External (TPSA, FieldUnitObj)    // (from opcode)

    Scope (_SB.PCI0.I2C0.TPL0)
    {
        Method (_INI, 0, NotSerialized)  // _INI: Initialize
        {
            If (_OSI ("Darwin")){}
            ElseIf (LLess (OSYS, 0x07DC))
            {
                SRXO (GPLI, One)
            }

            Store (GNUM (GPLI), INT1)
            Store (INUM (GPLI), INT2)
            If (_OSI ("Darwin"))
            {
                SHPO (GPLI, One)
            }
            ElseIf (LEqual (SDM0, Zero))
            {
                SHPO (GPLI, One)
            }

            Store (THHA, HID2)
            Store (TPSA, BADR)
            Store (0x00061A80, SPED)
        }

        Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
        {
            If (_OSI ("Darwin"))
            {
                Return (ConcatenateResTemplate (SBFB, SBFG))
            }
            Else
            {
                If (LLess (OSYS, 0x07DC))
                {
                    Return (SBFI)
                }

                If (LEqual (SDM0, Zero))
                {
                    Return (ConcatenateResTemplate (SBFB, SBFG))
                }

                Return (ConcatenateResTemplate (SBFB, SBFI))
            }
        }
    }
}