kecinzer/hpelitebook850g5-opencore #28: Cannot boot from USB Stick

ema07cqc, 2020-10-07:
Hi there. Its not an issue I think but i cannot boot from the USB Stick.
EliteBook 840 G5 here. Intel Core i5 8th Gen.
Updated and configured Bios (Q78 Ver 01.12.00) following provided instructions.

Created USB with Catalina Installer (SanDisk Cruzer Blade 16GB) and EFI Partition.
Copied entire EFI folder downloaded from this Git to the EFI Partition in the USB Stick.

When I try to boot from SanDisk USB, nothing happens.

There is a step missing?
Thanks.

[image]

[image]

[image]

Ive installed other hakintosh in the past, like Mojave using Clover on another EliteBook but i never used OpenCore.
Thank you very much.

labrouss, 2020-10-07:
Hi,

Try selecting boot from file and then select the usb stick -> efi -> opencore.efi file 

I also have this issue because I also need to press F9 then select boot from file etc ...

ema07cqc, 2020-10-07:
> Hi,
> 
> Try selecting boot from file and then select the usb stick -> efi -> opencore.efi file
> 
> I also have this issue because I also need to press F9 then select boot from file etc ...

Hi there. Tried that and same result.
With other EFI folders downloaded from Youtube I can get into the GUI but those EFI doesn't work because is for other models.

kecinzer, 2020-10-08:
This is weird.
I have one USB stick in FAT32 format withou EFI partition and it works for me also with right structure.
From you screens it looks good to me.
I think it will be some little think.

ema07cqc, 2020-10-08:
> This is weird.
> I have one USB stick in FAT32 format withou EFI partition and it works for me also with right structure.
> From you screens it looks good to me.
> I think it will be some little think.

Thanks for your reply. Downloaded files again, tried with another USB Stick (EFI partition and simply a FAT partition) and still I cannot get to the GUI. Maybe my EliteBook 840 G5 is different. Model No: 3RF20LT#AC8. Intel Core i5 8250U.
Have a nice day.

closed 2020-10-08 by ema07cqc