kecinzer/hpelitebook850g5-opencore #29: Touchpad files - doubt

Joaotcs, 2020-10-17:
Hey!
I have seen other works that, for touchpad (and touchscreen) they only use this kind of files, with no renames:

DefinitionBlock ("", "SSDT", 2, "hack", "_TPD0", 0x00000000)
{
    External (_SB_.PCI0.I2C0.TPD0, DeviceObj)    // (from opcode)

    Scope (_SB.PCI0.I2C0.TPD0)
    {
        If (_OSI ("Darwin"))
        {
            Name (OSYS, 0x07DC)
        }
    }
}

Why is it better the XINI and XCRS rename than this?

kecinzer, 2020-10-18:
This doesn’t work for me.
You can try it.
If this work it will be the best solution of course :).

Joaotcs, 2020-10-18:
It works for me...
Try it:

`/*
 * Intel ACPI Component Architecture
 * AML/ASL+ Disassembler version 20180427 (64-bit version)(RM)
 * Copyright (c) 2000 - 2018 Intel Corporation
 * 
 * Disassembling to non-symbolic legacy ASL operators
 *
 * Disassembly of iASL6uILaG.aml, Sun Oct 18 14:36:17 2020
 *
 * Original Table Header:
 *     Signature        "SSDT"
 *     Length           0x00000067 (103)
 *     Revision         0x02
 *     Checksum         0x95
 *     OEM ID           "hack"
 *     OEM Table ID     "_CPD0"
 *     OEM Revision     0x00000000 (0)
 *     Compiler ID      "INTL"
 *     Compiler Version 0x20180427 (538444839)
 */
DefinitionBlock ("", "SSDT", 2, "hack", "_CPD0", 0x00000000)
{
    External (_SB_.PCI0.I2C1.CPD0, DeviceObj)    // (from opcode)

    Scope (_SB.PCI0.I2C1.CPD0)
    {
        If (_OSI ("Darwin"))
        {
            Name (OSYS, 0x07DC)
        }
    }
}`

kecinzer, 2020-10-19:
OMFG!! @Joaotcs its actually working! 
I spent months to find this kind of solution but without success!
How you find it?

Would you like to create push request with these changes, or I can make it with your mention.

Thank you!

Joaotcs, 2020-10-19:
It's OK for me, but as I have some modifications, I will fork it now, and upload my changes.
Glad I could help ;) you sure did help me 😉

Thank you for sharing and constantly updating!

Now the how's.. Well, I have seen some works for other laptops that used this kind of aml files, but for different devices. I then modified and created new ssdt adapted to our laptops. In fact, I made first for the probook 440 g5, which uses another touchpad, and it worked. So I tried to make for this one. It also worked, so I created this topic because I didn't understand why the need for this renames :p

Joaotcs, 2020-10-19:
The code is getting pretty clean. I like it 😎👌

closed 2020-12-04 by kecinzer