kecinzer/hpelitebook850g5-opencore #32: Bluetooth 5.0

Rixxx-git, 2020-10-27:
Hi guys,

I use a Speedlink VIAS Nano 4.0 Bluetooth USB Adapter instead if a build in bluetooth card. Now I got some Airpods (2. Gen) and they don't show up, when I search for them. The Speedlink Adapter works with all my other bluetooth devices (iphone, huawei phone etc.). 

Can it be, that it doesn't recognize the Airpods because they use bt 5.0? 

And if so, how do I get 5.0 without changing the bluetooth/wifi card?

Thank you for your help! 🙂

kecinzer, 2020-10-27:
I'm using Fenvi card with BT 4.0. I have AirPods Pro and it works without problem. It should work for you.
When I connect Airpods Pro to my iPad Air 2019 that has BT 5.0 I have only better coverege.

Rixxx-git, 2020-10-27:
Thanks for your quick reply but unfortunately it's still not working. The airpods work with my Huawei Phone but they don't show on my Mac. 😔

Rixxx-git, 2020-10-27:
Got it. Had to press and hold the button on the back side. 🙂