kecinzer/hpelitebook850g5-opencore #34: Almost worked for 840g5. Thanks!

kmasterycsl, 2020-11-03:
I am just a beginner in hackintosh field. But with your effort, i can make it work on my 840g5 too 👍 . Except the replacement for the weird PM 981 SSD.
I try OpenIntelWireless with my original intel wlan card and it works. You can try it too 😄 
Thanks!

Rixxx-git, 2020-11-05:
You mean wlan works with your integrated card? 😍 does bluetooth work as well?

Wizeguy24, 2020-11-05:
Intel wireless and Bluetooth working for me as well from OpenIntelWireless team with 850G4 my side

Rixxx-git, 2020-11-05:
Wow, how did you do this exactly? 😲

Wizeguy24, 2020-11-05:
All the install instruction are on the below links, the simple version is download kexts, add to kexts folder, update config.plist. 

https://github.com/OpenIntelWireless/IntelBluetoothFirmware
https://github.com/OpenIntelWireless/itlwm

Rixxx-git, 2020-11-05:
Thanks! Unfortunately there is no instruction for the config.plist changes. Can you please let me know which changes both the wifi and the bluetooth kexts need? 🙂

Wizeguy24, 2020-11-05:
Put the kexts in the kext folder, there will be 3:
AirportItlwm.kext (or itwm.kext)
IntelBluetoothFirmware.kext
IntelBluetoothInjector.kext

Open your config.plist with ProperTree and go to File then  "OC Snapshot" it will add them for you

Rixxx-git, 2020-11-05:
Thanks again!

I opened the config.plist with ProperTree, but when I go to file and "OC Snapshot" I can't choose the itwm.kext file. (see screen) Or which file do I have to choose?

[image]

Rixxx-git, 2020-11-05:
Do I have to choose the OC Folder in my EFI folder? 🙂

labrouss, 2020-11-05:
Hi,

Works for me as well. Still haven’t been able to make handoff and airdrop to work though. Has anyone solved that ?

Wizeguy24, 2020-11-05:
> Do I have to choose the OC Folder in my EFI folder? 🙂

Yes, it will add it then. Always best practice to test on a USB first and boot off there so you don't damage existing install.
Just safer that way.

Rixxx-git, 2020-11-05:
Unbelievable, it works! Thank you guys! I can't use the bluetooth kext because my card is x8086 instead of x8087, but wifi is already awesome. 🙂 Any experiences with a x8086 card? Does it work as well?

Rixxx-git, 2020-11-06:
Edit: the bluetooth card seems to be always x8087 for my laptop model. So I added the kexts and now bluetooth is working fine as well. Thanks so much again guys! No more dongles! 😄

kecinzer, 2020-11-06:
Do I understand it correctly, that with those kexts Intel 8265NGW wifi card works well?
It will be great if Intel AX201 works because of BT 5.1.

Wizeguy24, 2020-11-06:
Intel Dual Band Wireless-AC 8265 802.11 a/b/g/n/ac (2x2) Wi-Fi and Bluetooth 4.2 combo,  definitely works thats what i'm using. 

Looks like AX201 is working as well: https://www.reddit.com/r/hackintosh/comments/he9x31/how_is_the_intel_wifibt_support_going/

Rixxx-git, 2020-11-06:
Intel Dual Band Wireless-AC 8265 802.11 a/b/g/n/ac (2x2) Wi-Fi and Bluetooth 4.2 combo works for me as well with those kexts. 🙂

kecinzer, 2020-11-06:
Thank you for info. I already red about it on project wiki. It’s still no go for me as I have fully supported Fenvi M2 card. But this card is AC/BT4. It works great for now, but I would like to switch to Wifi AX/BT 5.x in future.

archfish, 2020-11-09:
How do you configure the touchpad properties? Trackpad setting showing "no trackpad found". 

And my Thunderbolt 3 port can not output screen.

840 g5 with BIOS version 1.14.01.

kecinzer, 2020-11-09:
Working normaly for me.
[image]

archfish, 2020-11-10:
> Working normaly for me.
> [image]

Is there any document let me to troubleshoot this issue? 
Thanks in advance.

archfish, 2020-11-10:
Using new master EFI,trackpad work normaly. Thanks!!

closed 2020-12-04 by kecinzer

GoForce5500, 2020-12-11:
Why this just can't boot on my 840g5, it always return to BIOS. And ProperTree will complain about config.list.

kecinzer, 2020-12-11:
May be due to some new security stuff.