kecinzer/hpelitebook850g5-opencore #35: Latest bios breaks battery

Joaotcs, 2020-11-08:
With latest bios 1.14, battery is not recognized in win10 nor macos. And in win10 its really laggy. Back to previous version for now

Joaotcs, 2020-11-08:
Warning, can't rollback bios...

Joaotcs, 2020-11-08:
Will need your help debugging this

Joaotcs, 2020-11-08:
It seems that blocks downgrades AFTER you either activate HP SURE or return to factory settings...

After bios, my DSDT patch gives me this errors:
```

// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECMX], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x1db (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECRG], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x1f1 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.PSCF], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x207 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.BCSC], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x21d (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^RP01.PXSX], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x7f7 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^^IETM.DFFG], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x80c (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECMX], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x1db (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECRG], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x1f1 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.PSCF], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x207 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.BCSC], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x21d (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^RP01.PXSX], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x7f7 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^^IETM.DFFG], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x80c (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECMX], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x1db (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECRG], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x1f1 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.PSCF], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x207 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.BCSC], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x21d (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^RP01.PXSX], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x7f7 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^^IETM.DFFG], AE_NOT_FOUND (20180427/dswload-498)
// // Invalid external declaration at AML offset 0x80c (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECMX], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x1db (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.ECRG], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x1f1 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.PSCF], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x207 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^PCI0.LPCB.EC0.BCSC], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x21d (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^RP01.PXSX], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x7f7 (see bz1397).
// Firmware Error (ACPI): Failure looking up [^^^IETM.DFFG], AE_NOT_FOUND (20180427/dswload2-468)
// // Invalid external declaration at AML offset 0x80c (see bz1397).
```

So it seems that we have to change the ssdt-bat.aml in the section "    Scope (\_SB.PCI0.LPCB.EC0) "

Will have to study this, but I ain't got time the needed time for now
Might be an opportunity to try to clean the renames codes...

kecinzer, 2020-11-08:
Ouch, I can see new BIOS 6.11.20. So I will stay with current version until we found solution for this...

Joaotcs, 2020-11-08:
It seems that the lagginess was software related.
The only problem is battery not being recognized In Mac (in windows it works). 
This new bios fixes a lot of bugs, and updates almost everything bios-related. 
Any advice on how to proceed on fixing this? I was thinking in having a DSDT of the previous bios and compare to my computer now... What do you think? 

You may try it, BUT DO NOT RESET FACTORY SETTINGS (because it activates HP Sure). If you activate ONCE HP sure, rollback is impossible. If you just don't reset factory settings you can go back. Don't make my mistake.

Joaotcs, 2020-11-08:
I won't have time for the next two weeks, due to covid, but it seems that using this two guides, it should be easy to make battery work again:
https://www.tonymacx86.com/threads/guide-how-to-patch-dsdt-for-working-battery-status.116102/

https://www.tonymacx86.com/threads/guide-using-clover-to-hotpatch-acpi.200137/

I may even ask for help to this guy: https://www.tonymacx86.com/threads/opencore-battery-patch.295289/

kecinzer, 2020-11-08:
I know about this guides, but I don’t have much time now. Battery hotpatching seems to be hard and long process.
Did you tried remove ACPI renames? I think you can compare unmodified DSDTs.

kecinzer, 2020-11-08:
I looked at it and I think I can fix it. I will try it tomorrow. I suppose there was some regions change that need to be fixed in SSDT-BATT.dsl.

Joaotcs, 2020-11-09:
Yes, I also don't have time for now, but it's ok, I plug the computer from time to time... And If I boot to windows bypassing opencore patches, I can see the battery anyway.

Had to warn here the rollback problem anyway because you seem to update everytime :p

kecinzer, 2020-11-09:
Partially successful...
[image]

Joaotcs, 2020-11-09:
That's great! Battery is at least recognized now ;)

mehmetyukcell, 2020-11-09:
Use SMCBatteryManager instead of ACPIBatteryManager.

Joaotcs, 2020-11-09:
We are using SMCBM. did you mean the contrary?

kecinzer, 2020-11-09:
Well, fixed :)
https://github.com/kecinzer/hpelitebook850g5-opencore/commit/596df8ba09c256e14e18073f4c59199cb38c7e11

Joaotcs, 2020-11-09:
God-like !!!
Many thanks ;) Closed topic.

closed 2020-11-09 by Joaotcs

kecinzer, 2020-11-09:
Almost all day of fun :). I’m glad that I learned new things.