kecinzer/hpelitebook850g5-opencore #37: Trackpad - VoodooI2C intermittently laggy/non responsive

luka-diplomski, 2020-11-13:
Has anybody noticed trackpad multi touch gestures being less responsive somewhat? 
I dont know how to describe but Ill try. If you use virtual desktops and 3-finger gestures to swipe between them I noticed that maybe the first 3 times I swipe it works then the forth time it doesn't work for cca 2secs. Like it ignores all action in this short time period. I will make a video later.

I dont know when I first started noticing this, but I think around Big Sur RC2. I first thought it is in my head but then I booted Catalina latest and noticed issue is non existent there.

kecinzer, 2020-11-13:
I will test it and let you know...
Recently I start using Magic Mouse :).

kecinzer, 2020-11-13:
It works normally for me.

luka-diplomski, 2020-11-14:
Thanks for checking out. Now it works for me perfectly. When starts to happen again, after I reboot it works again. I will check the logs.

I will close this.

closed 2020-11-14 by luka-diplomski