kecinzer/hpelitebook850g5-opencore #38: ROM error on startup

Rixxx-git, 2020-11-14:
Hi guys, I get this error on startup:
OCS: Failed to parse data field as value with type data and <"my mac address"> contents, context <ROM>!

The number in the "my mac address" field is my real mac address from network preferences and I use it as the ROM value in my config.plist. My Mac starts and works fine. I just wanted to know why I get this error and what I can do about it?

Thank you all!

kecinzer, 2020-11-14:
Please double check you rom address format.

Rixxx-git, 2020-11-14:
[image]
[image]

Is the format in my config.plist wrong?

labrouss, 2020-11-14:
It’s not the MAC address itself. It’s calculated from the MAC. I use clover to calculate SMBIOS values 

Kind regards,
Konstantinos

labrouss, 2020-11-14:
I mean clover configurator

Kind regards,
Konstantinos

Rixxx-git, 2020-11-14:
you mean like this?

[image]

I still get the same error with this ROM value

Rixxx-git, 2020-11-14:
got it. exported the config.plist from clover editor. the rom value has now only 8 characters.

closed 2020-11-14 by Rixxx-git

labrouss, 2020-11-14:
Yes like that ! 

If you are migrating from a CLOVER boot loader you might as well copy the smbios from CLOVER. 

Otherwise generate new SMBIOS settings using clover configurator