kecinzer/hpelitebook850g5-opencore #40: Battery Icon doesn't show in Menu Bar / Control Center

Rixxx-git, 2020-11-22:
Hi everybody,
I recognized a few days ago, that the battery icon doesnt show in the menu bar or control center. I enabled both in the settings.

I use your latest EFI plus Wifi & Bluetoothfirmware kext to use the internal card. My BIOS version is 1.11. or 1.12. I didn't update to the latest version. Might that be the reason or is it a Big Sur issue?

Thanks for your help!

Axon80, 2020-11-22:
Hi,
You need to update your bios to the latest version.

Rixxx-git, 2020-11-22:
Will there be no other issues when I update the BIOS version on my 830 G5?

Axon80, 2020-11-22:
The author uses 850 G5 and has adapted SSDT for this latpoa. You are using the hp 830 G5. I don't know if it will work ..... Maybe you need to go back to the old SSDT ....

Rixxx-git, 2020-11-22:
@kecinzer What do I have to change about your setup for my 830 G5? 🙂  I think not the bios but the latest ssdt update is the reason.

Wizeguy24, 2020-11-22:
> @kecinzer What do I have to change about your setup for my 830 G5? 🙂 I think not the bios but the latest ssdt update is the reason.

I had the same issue as you on my 850 G4 - HP released a Bios for mine at the same time as the G5. Updated Bios and all worked perfectly again. I uses Intel Wifi and Bluetooth kexts as well.

Rixxx-git, 2020-11-22:
Great, thank you! I will try it.

Rixxx-git, 2020-11-22:
I just updated the BIOS to 1.14. but the icon is still not there. It doesn't even show in the settings sidebar. Only if I search in the settings for "Battery".

[image]

[image]

Rixxx-git, 2020-11-22:
@Wizeguy24 Now Heliport is not working anymore. I can't turn it on. :(

kecinzer, 2020-11-22:
May be there are some other differences in 830 G5. Did you updated SSDT-BATT.aml and also patches in config.plist? I changed lot of things according to battery. Mainly ACPI renames and battery ACPI hotpatch.

igorkolesnikov13, 2020-11-23:
@Rixxx-git, just for your information. I have 840 G5, and after installing the latest BIOS and using the latest config battery is working flawlessly.

Rixxx-git, 2020-11-23:
Got the battery icon. Forgot to update the config.plist. 

But Heliport is still not working after the update to latest BIOS. @Wizeguy24 I added all 3 kexts and updated config.plist. Any idea?

@kecinzer Why did you change from Macbook15,4 to 14,1 in config.plist? 🙂

[image]

[image]

kecinzer, 2020-11-23:
@Rixxx-git I'm so sorry, but as I use native Wifi card, I don't support any Intel kexts or drivers. You should ask on that project.

I switched to MacBookPro14,1 according to reading of many threads and also my performance tests. MacBookPro14,1 is better choice in performance. 14,1 also don't have touchbar or touchid.

Rixxx-git, 2020-11-23:
@kecinzer Don't be sorry! You are doing an awesome job here! 🙂  @Wizeguy24 seems to use your EFI setup, latest BIOS and the Intel kexts. So maybe he can tell me what to do. 🙂

Wizeguy24, 2020-11-23:
I use the airportitlwm for big sur which uses native wifi controls, no need
for heliport then.

https://github.com/OpenIntelWireless/itlwm/releases

Rixxx-git, 2020-11-23:
Thanks. I changed itlwm kext to airport kext and changed securebootmodel to default in config.plist like it says in the description. 

Now my laptop doesn't start anymore. Secure boot should stay disabled right? How can I change that again if the laptop doesn't start?

kecinzer, 2020-11-23:
Disable secureboot.

Rixxx-git, 2020-11-23:
It is disabled in bios settings. How can I disable it again in my config.plist if it doesn't start?

Wizeguy24, 2020-11-23:
Boot from usb with EFI on it. Should always have an install usb with EFI
for recovery.

Rixxx-git, 2020-11-23:
Got it running again. Thanks so much! :)

WiFi is not finding any network and Bluetooth still doesn't show in the settings. I use Bluetooth firmware and inject kext. And it was working before Bios update.

Wizeguy24, 2020-11-23:
Only 2 options i can think of now is to check for newer version of heliport
and itlwm kext for big sur or post on their github.

Rixxx-git, 2020-11-23:
@Wizeguy24 Do I have to use itlwm, airportlwm, bluetoothfirmware and injector all together?

Rixxx-git, 2020-11-23:
@kecinzer You have this in your config.plist: 
<dict>
				**<key>Arch</key>
				<string>x86_64</string>**
				<key>BundlePath</key>
				<string>Lilu.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/Lilu</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>

When I create a config.plist with my OC folder and ProperTree it doesn't create those bold lines:

<dict>
				<key>Comment</key>
				<string></string>
				<key>MaxKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
				<key>Enabled</key>
				<true/>
				<key>MinKernel</key>
				<string></string>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/Lilu</string>
				<key>Arch</key>
				<string>Any</string>
				<key>BundlePath</key>
				<string>Lilu.kext</string>
			</dict>

Does that make any difference?

Wizeguy24, 2020-11-23:
itlwm and heliport go together for wifi or airportlwm only.

Bluetooth files go with either option above.

Rixxx-git, 2020-11-23:
Strange. itlwm and heli don't work. airportlwm alone doesn't work as well for me.

Wizeguy24, 2020-11-23:
Are you doing a clean snapshot in propertree or manually adding in ?

Rixxx-git, 2020-11-23:
I choose "OC Snapshot" and then the OC folder. After that I add the code created in ProperTree to the kext part of the config.plist of @kecinzer. 

Should I use "OC Clean Snapshot"?

Maybe you can show my your config.plist without all the Serial Numbers? 🙂

Joaotcs, 2020-11-23:
All G5 are the same. Maybe USB ports are different. The rest, is the same!!
Even G4 seems to be compatible....
G6 Audio is different, the rest is also the same.

I think if you solved your problem you can close the topic ;)

Rixxx-git, 2020-11-23:
> All G5 are the same. Maybe USB ports are different. The rest, is the same!!
> Even G4 seems to be compatible....
> G6 Audio is different, the rest is also the same.
> 
> I think if you solved your problem you can close the topic ;)

Right. The battery problem is solved, but now internal wifi & bluetooth doesn't work anymore.

DominikHackintosh, 2020-11-23:
Rixx-git try the alpha relase from itlwm, i only use this one https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha

Rixxx-git, 2020-11-24:
> Rixx-git try the alpha relase from itlwm, i only use this one https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha

Thanks! You use itlwm alpha with Heliport or airportitlwm alpha as standalone?

DominikHackintosh, 2020-11-24:
Only airportitlwm 😎

Rixxx-git, 2020-11-24:
> Only airportitlwm 😎

I added the big sur alpha version but it still can't find any network:

[image]

Rixxx-git, 2020-12-04:
@kecinzer I installed the fenvi wifi card. All working fine now. 🙂 But I changed my EFI to the one from @Joaotcs 
Which part of your EFI do I need to solve the Battery Icon problem? It doesn't show again for me now.

Joaotcs, 2020-12-04:
My shared EFI is not updated to the latest BIOS, will update today.
Without the latest security modification from kecinzer

Joaotcs, 2020-12-04:
> @kecinzer I installed the fenvi wifi card. All working fine now. 🙂 But I changed my EFI to the one from @Joaotcs
> Which part of your EFI do I need to solve the Battery Icon problem? It doesn't show again for me now.

Updated now. Please see if it works now ;)
Regards

closed 2020-12-04 by Rixxx-git