kecinzer/hpelitebook850g5-opencore #41: Rollback Bios

Rixxx-git, 2020-11-23:
Hi guys,

as the new bios version 1.14. doesn't work for me on my 830 G5 with intel kexts for internal wifi & bluetooth, I would like to try and roll back to the previous bios version: 1.12.0.0

When I start the rollback in bios it says that "System Administrator has locked bios version".

I didn't lock the version in bios settings. Is it somehow locked in the EFI files?

DominikHackintosh, 2020-11-23:
Check in bios, there is an option. Uncheck it

Rixxx-git, 2020-11-23:
It is unchecked.

kecinzer, 2020-11-23:
BIOS can’t be locked by EFI files.
I really recommend you to follow [THIS GUIDE](https://dortania.github.io/Wireless-Buyers-Guide/types-of-wireless-card/m2.html#supported-cards) and buy fully supported wifi card. I don’t use Intel Wifi nor BT, so I don’t support it. May be in future when drivers will be stable and support all what I need (Handoff, AW unlock).
I’m early adopter and I will upgrade to latest BIOS ASAP it comes out. Don’t expect from me that I ever will roll back to older version :)).

DominikHackintosh, 2020-11-23:
Hi kecinzer, 

the itlwm project was growing in the last time and extremely successfully. I was asking a long time to change my card, but now definitely no more, handoff, airdrop so far is working. really great and now also all older centrino cards are working. just as info, i am really happy with it, is use only the prerelease alpha and update nearly every day. greeetings from hungary

kecinzer, 2020-11-24:
> Hi kecinzer,
>
> the itlwm project was growing in the last time and extremely successfully. I was asking a long time to change my card, but now definitely no more, handoff, airdrop so far is working. [...]

I red few weeks ago, that is impossible to connect to enterprise network (secured by certificate). Speed is only in N not AC or X.
I'm fan of this project, because there is no BT 5.x official card for hackintosh and I want it in the future.

But in fact, now when Apple switching to Apple Mx SOCs I think I will switch to Apple HW before I need to change my Wifi/BT card.

Joaotcs, 2020-11-24:
If you clicked on go factory setting or something enabled the HP Sure, once you do any of this, you are locked to rollback to previous version. This is written in the changelog, and I also warned elsewhere in this topics.

So you cannot downgrade atm.
itlwm is not stable, i have tried it, it is much better to spend 40eur to just buy a fully working card with broadcoam chip.

Joaotcs, 2020-11-24:
Yap, it seems ARM is the future. So I give hackintosh 3 years, until it's dead - or at least, non upgradable anymore.
ARM is much more efficient, the only problem is that most of the software is not optimized, but there will also be an ARM alternative from Intel I am sure, as it's the right path atm.

Performance wise, for a first "beta" SOC, M1 beats the crap out of at least any i5 CPU intel has, but it needs an optimized app. As they take time to optimize, you are either going to get a slower performance for non optimized apps (or non working at all), but with optimized apps, you will see a much better performance vs Intel X86 cpus. Intel is going down the hill soon. Qualcomm will probably take place as the king of CPUs soon.

Btw, @kecinzer , what is you "expected time" with this computer before switching ? A year, 2y, 3y tops?

Rixxx-git, 2020-11-24:
> If you clicked on go factory setting or something enabled the HP Sure, once you do any of this, you are locked to rollback to previous version. This is written in the changelog, and I also warned [...]

Thank you! At least I know why it's not working.

Rixxx-git, 2020-11-29:
The BIOS settings say I can rollback to a Version if I insert it on an USB Stick. Is it possible to download the 1.12. BIOS Version somewhere? I can't find it.

labrouss, 2020-11-29:
I was also having issues and now using another fork. 

Kind regards,
Konstantinos

Rixxx-git, 2020-11-29:
Thanks! Which fork do you use? Did you update to the latest BIOS version (1.14.) ?

Rixxx-git, 2020-11-29:
> I was also having issues and now using another fork. Kind regards, Konstantinos
> […](#)

Found your fork. Do you use the latest BIOS version?

Joaotcs, 2020-11-29:
Hey, man. You can afford a 1200eur but not 25eur for adapter + apple card or 45 for fenvi card?? I think it's worth enough the trouble.. Much more reliable.

Anyway, I will repeat, this latest bios BLOCKS downgrade IF you did bios reset OR had HP sure enabled. As you already say you cannot downgrade I am certain that you already did this. So, 
NO, YOU CANNOT DOWNGRADE. This bios updates a lot of stuff, and it's written on hp changelog, that if you do bios reset to default setting or enable hp surely, you are stuck with this bios.
So please, don't bother trying to find a way to downgrade

Joaotcs, 2020-11-29:
I have bought the fenvi card. I can sell you the apple+adapter card, 20eur+shipping of you want..

Rixxx-git, 2020-11-29:
> Hey, man. You can afford a 1200eur laptop but not 25eur for adapter + apple card or 45 for fenvi card?? I think it's worth enough the trouble.. Much more reliable.
> 
> Anyway, I will repeat, this latest bios BLOCKS downgrade IF you did bios reset OR had HP sure enabled. As you already say you cannot downgrade I am certain that you already did this. So,
> NO, YOU CANNOT DOWNGRADE. This bios updates a lot of stuff, and it's written on hp changelog, that if you do bios reset to default setting or enable hp surely, you are stuck with this bios.
> So please, don't bother trying to find a way to downgrade

I already bought a broadcom card last year but then somebody reported problems when integrating it with the older clover setup. So I decided to sell it again. It's not about the money. 😉

How does your apple card work? Do I have to open my laptop and integrate it?

Rixxx-git, 2020-11-29:
Also: There are some people using the latest BIOS and they don't have any problem using the internal intel wifi card with the kexts. So I thought I might did some other mistake.

labrouss, 2020-11-29:
I’m one version back from the latest and don’t plan to upgrade. I can find it but first give the following a try 

I’m using the following which I like better due to open canopy picker 

https://github.com/Joaotcs/Hackintosh-Elitebook-8x0-g5

Kind regards,
Konstantinos

Rixxx-git, 2020-11-29:
> I’m one version back from the latest and don’t plan to upgrade. I can find it but first give the following a try I’m using the following which I like better due to open canopy picker https://github.com/Joaotcs/Hackintosh-Elitebook-8x0-g5 Kind regards, Konstantinos
> […](#)

Thanks. Tried that fork already but it's still not working. So better don't update your BIOS.

For itlwm kext Heliport can't be turned on. Airportitlwm doesn't find any network.

Rixxx-git, 2020-11-29:
>  I can find it but first give the following a try 

You think you can find the 1.12. BIOS Version? Although @Joaotcs thinks it's not possible, I would like to try and rollback with the version on USB.

Joaotcs, 2020-11-29:
> > I can find it but first give the following a try
> 
> You think you can find the 1.12. BIOS Version? Although @Joaotcs thinks it's not possible, I would like to try and rollback with the version on USB.

Been there, done that. You are stubborn.. Do it as you wish. You Will waste time. Buying the fenvi card or Broadcom is 100% compatible, no problems, stable. But be my guest and try wiuth USB.
Just read the hp changelog of 1.14 bios. The red letters.

Rixxx-git, 2020-11-29:
Thanks. I didn't know you tried that as well. Which apple card do you sell? 🙂

Joaotcs, 2020-11-29:
Bcm94360cs2.
It's either that or fenvi.

Rixxx-git, 2020-11-29:
What is the difference between those 2 cards?

Joaotcs, 2020-11-29:
Fenvi is double the price but doesn't need adapter. Apple card needs adapter and its harder to fit.
But they are the same card basically

Rixxx-git, 2020-11-29:
Ok thanks. Is that one right: https://www.amazon.de/-/en/gp/product/B083YXS7VF/ref=ox_sc_act_title_1?smid=A3OJWAJQNSBARP&psc=1 ?

Joaotcs, 2020-11-29:
Yap, that's the fenvi one

Rixxx-git, 2020-11-29:
Thanks. Will order it. 🙂

closed 2020-11-29 by Rixxx-git