kecinzer/hpelitebook850g5-opencore #42: Authenticated Root is active

Rixxx-git, 2020-11-25:
Since a few days I get this notice, when I run kext updater. Is that right?

[image]

kecinzer, 2020-11-25:
What is your current csr-active-config value? I use 00000000, so csrutil is completly enable for full security. I never had issues with it.

Rixxx-git, 2020-11-25:
I never changed the value. From your setup it has been AAAAAA== all the time. 🙂

kecinzer, 2020-11-25:
And what is output, when you type "csrutil status" in Terminal?

Rixxx-git, 2020-11-25:
I set it to the value in the picture and I also tried 000000. But still get the info that root is active. 

Terminal says: System Integrity Protection status: enabled.

[image]

kecinzer, 2020-11-25:
And why you need this? You have kexts somewhere else than in EFI folder? That is not supported for this project. I updating all kexts regularly.

Rixxx-git, 2020-11-25:
I don't need it. I was just curious what that root information is about. 🙂 I thought it might be important.
Thank you for your help!

closed 2020-11-25 by Rixxx-git