kecinzer/hpelitebook850g5-opencore #43: Itlwm & Airportitlwm Kexts for Intel Wifi & Bluetooth

Rixxx-git, 2020-11-25:
Although @kecinzer is not using the default Intel Wifi & bluetooth card I would like to discuss options how to be able to use it.

Lately there are those new kexts: https://github.com/OpenIntelWireless/itlwm

Before I updated my HP 830 G5 to the latest BIOS (1.14.) the Intel wifi & bluetooth was working fine with the stable version of Itlwm kext and Bluetoothfirmware + Bluetoothinjector kext. After the BIOS update Heliport, which is needed for Itlwm to work, can't find any networks and my Bluetooth settings disappeared from system preferences. 

I also tried Airportitlwm but it can't find any network as well.

Any help is greatly appreciated! 🙂

DominikHackintosh, 2020-11-25:
try to delete your complet networksettings Library/Preferences/SystemConfigurations/....

Rixxx-git, 2020-11-25:
What exactly should I delete in this folder? Everything?

Rixxx-git, 2020-11-25:
Do I have to clear NVRAM and Kext cache always after making changes to config.plist or kext folder?

Rixxx-git, 2020-11-25:
@DominikHackintosh Can you post your config.plist so that I can see how you added Itlwm Kexts to it? :)

DominikHackintosh, 2020-11-25:
you will find informations for that in my repo

Rixxx-git, 2020-11-25:
You write this in your repo: NOW all new BIOS Versions are working! ((((Update/Downgrade!!! to BIOS P78 v.1.29 / v.1.3.1 (extremely important becaus newer Versions don´t work!!!!))))

So it has to be about the BIOS update that I did and I can't roll back...

Rixxx-git, 2020-11-25:
I wrote this in terminal to see if the kexts load properly: kextstat | grep -i "IntelBluetoothFirmware"

and got this:
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release

What does it mean?

Rixxx-git, 2020-11-29:
Heliport will not turn on. This is my log file:
[HeliPort_logs.log](https://github.com/kecinzer/hpelitebook850g5-opencore/files/5612176/HeliPort_logs.log)
[itlwm_logs.log](https://github.com/kecinzer/hpelitebook850g5-opencore/files/5612177/itlwm_logs.log)

I did a reset of the Network System Preferences but still nothing.

Intel Wireless 8265 / 8275

closed 2020-12-04 by Rixxx-git