kecinzer/hpelitebook850g5-opencore #46: Hibernation (hibernatemode 3 or 25)

kecinzer, 2020-12-07:
I recently try to enable hibernation - sleep to disk but without success. I tried lot of things. HibernationFix.kext and many options, but it still not working.
After I put laptop to sleep and then wakes it, I can see different progress bar, but right after I can se cursor, it will reboot.
Is there someone who have working hibernation?

Wizeguy24, 2020-12-08:
Not tested yet but i see there is a HibernationFixup kext. https://github.com/acidanthera/HibernationFixup

kecinzer, 2020-12-08:
Did you read my post? I mentioned that I tried this kext without success 😞.

Wizeguy24, 2020-12-08:
apologies, multi-tasking while in a boring meeting lol

DominikHackintosh, 2020-12-10:
i have hp 840g4, was struggling around a long time with sleep. now everything works great. did you try with hackintool or kextupdater those option to set hiberfile to zero and write protected. this helped me alot
[image]
[image]

igorkolesnikov13, 2020-12-11:
@DominikHackintosh hi. I think that the thread is about to enable hibernation (sleep to disk) when the laptop won't use any power. The fix that you presented (is a really useful one) only refers to disabling hibernation and making the laptop sleep to RAM (with uses power, something like 2% per 8 hours).
Realy hibernation is really exciting!

DominikHackintosh, 2020-12-11:
i am sorry did not read carefully that he means sleep to disk. for me completely new, did not know that mac os has this option