kecinzer/hpelitebook850g5-opencore #47: Can't find Big Sur install media on OC

Se7Touch, 2020-12-09:
Hello, I make a new Big Sur install media and copy efi files to flash disk.

I boot laptop from usb and see OC menu but can't find "install macOS" drive...

Have any idea ?

kanmissioner, 2020-12-09:
Did you format the drive in disk utility?

Se7Touch, 2020-12-09:
Yes, all function is okey with other efi files.

Its problem only this efi..

kecinzer, 2020-12-10:
Are you following Dortania Guide? I tested Win10 and now I will go back to macOS, so I can let you know how it going.

kecinzer, 2020-12-10:
Hmm, I created installer on Windows and can’t also see install option in OC. Weird.

kecinzer, 2020-12-10:
@Se7Touch figured it out. You need to set “ScanPolicy” to 0! Installing right now :).

closed 2020-12-10 by kecinzer

Se7Touch, 2020-12-11:
Yes, I fixed the problem but forgot to let you know.  Thank you :)