kecinzer/hpelitebook850g5-opencore #48: AC sleep problems

ziyilang2000, 2020-12-15:
When I use AC power ，Close the monitor，Laptop can't sleep，Keep making a sound, like BOOP，always. 
click sleep button, no problem. 
The problem only occurs when the monitor is turned off. There is no problem using the battery

Ermac-X, 2020-12-18:
i'm having the same problem ... the device can sleep with sleep option  from menu with AC Power ,but with closing lid i keep getting this beep and device doesn't sleep .. on battery everything works fine

kecinzer, 2020-12-18:
How long this happens? It's new issue after some commit? I never had this problem.

Ermac-X, 2020-12-18:
I was on Catalina for a while an update forced me to leave Clover and use your Opencore everything worked fine with big Sur upgrade , just this issue ... on both Catalina and Big Sur 
btw is there a way to get my touch screen work again ? i have the touch screen variant and it was working on Clover .. thanks in advance

kecinzer, 2020-12-18:
Touch screen muset be enable by similar ACPI patch https://github.com/kecinzer/hpelitebook850g5-opencore/blob/master/SSDT/SSDT-CPD0.dsl. May be @Joaotcs can help.

Joaotcs, 2020-12-18:
Use my tpl0.aml acpi patch and add it on config.plist

Yes, I also notice this bug with closing lid, it gets a sound like I am constantly pressing a Key... It is from newer commits. Maybe the input.aml cleaning did this. Will try the older version acpi patch.

Ermac-X, 2020-12-19:
@Joaotcs I Used your TPL0.aml acpi patch and added to to config.plist but no luck sadly or i did something wrong since i'm totally new to OC, i tried to boot from your own OC the trackpad never worked .. 
I hope you would manage to solve this annoying bug of sleep on AC ... you guys saved me from buying a Mac Book with your great effort ! aside from touch and webcam everything seems to work like a charm !

Joaotcs, 2020-12-19:
> @Joaotcs I Used your TPL0.aml acpi patch and added to to config.plist but no luck sadly or i did something wrong since i'm totally new to OC, i tried to boot from your own OC the trackpad never [...]

You have to use both cpd0 and tpl0... I haven't done any mod to trackpad, so you messed up something, as my trackpad is the same as kecinzer. Cpd0 is track pad. Tpl0 is touchscreen... You have to know which is the device of touchscreen.. It might be different then others.

Ermac-X, 2020-12-19:
weird that i have both in my OC but only trackpad works ... mine works on kecnizer OC i suppose i also have the same as yours both ... just no idea why is it didn't work with ur OC 
do you mean my touchscreen might be different and some additional changes needed ?

kecinzer, 2020-12-19:
Yes, we switched from XOSI universal rename to more elegant hack. But you need to know xour device name. May be you have TPL1 or something like that.

Ermac-X, 2020-12-19:
and here is my problem im totally out of the context of all those details

Ermac-X, 2020-12-19:
trying not to mess up a lot with my device as i depend on it for work and study .I tried to google somehow but being under work pressure and lack of any experience with those Patches made it like endless cycle for me .. it is bothering not to have the touch working though it isn't very important but hope at least the sleep issue could be fixed somehow ..trying not to mess up a lot with my device as i depend on it for work and study

Joaotcs, 2020-12-21:
Do you have a dump of your ACPI? Please use a older version of Kecinzer (before the touchpad change), and when you boot with that efi, if the touchscreen is working, open the app ioregistryexplorer, and write " TPL", and give me a screenshot, then, use the MaciASL, and give me the DSDT dump, or copy paste the corresponding lines of the "TPLX", being X the number of your device.
Regards

Ermac-X, 2020-12-21:
the only situation I got my touch screen working was on Catalina Clover which i was using for months and avoiding any update not to get into such troubles , till i mistakenly updated Catalina Security update which broke my touchpad and sleep.
was forced to move to OC and upgrade to Big Sur to follow up with you .. and eversince no touch screen ..
weird is that your kexts and ACPI are semi Identical to kecinzer yet with his i have fully functional touchpad and urs not 
tried to add TPL0 from yours to his and enable on OCC but it doesn't work

archfish, 2020-12-26:
Got same issue, after system reboot, first time close lid the audio keep beep.
system log showing :

```log
2020-12-27 00:01:24.072186+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.072197+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.072337+0800  localhost kernel[0]: (AppleIntelKBLGraphicsFramebuffer) <AppleIntelKBLGraphicsFramebuffer`AppleIntelFramebuffer::setAttribute(unsigned int, unsigned long)> [IGFB][LOG  ][DISPLAY   ] [Set_attribute] (attribute='bksv', value=58)
2020-12-27 00:01:24.077347+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.077357+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.077997+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.078008+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.081721+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.081732+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.082375+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.082390+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.084830+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.084839+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.085402+0800  localhost kernel[0]: IOConsoleUsers: time(1) 1608998482->1608998482, lin 1, llk 1,
2020-12-27 00:01:24.085411+0800  localhost kernel[0]: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 1608998484, sm 0x0
2020-12-27 00:01:24.087738+0800  localhost kernel[0]: (AppleCredentialManager) <AppleCredentialManager`deleteCredentialSet> ACM: deleteCredentialSet: Deleting credential set, CS[108].
2020-12-27 00:01:24.120237+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::externalMethod(unsigned int, IOExternalMethodArguments*, IOExternalMethodDispatch*, OSObject*, void*)> + IOAudioEngineUserClient::externalMethod, selector=0x0,   arg0 0x0, arg1 0x1019f5000, arg2 0x8010 arg3 0x6a
2020-12-27 00:01:24.120240+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::externalMethod(unsigned int, IOExternalMethodArguments*, IOExternalMethodDispatch*, OSObject*, void*)>   scalarInputCount=0x4  structureInputSize 0x0, scalarOutputCount 0x0, structureOutputSize 0x0
2020-12-27 00:01:24.120245+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)> + IOAudioEngineUserClient::registerBuffer64 0x0 0x1019f5000 0x8010 0x6a
2020-12-27 00:01:24.120248+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::safeRegisterClientBuffer64(unsigned int, unsigned long long*, unsigned int, unsigned int)> + IOAudioEngineUserClient::safeRegisterClientBuffer64 <private>
2020-12-27 00:01:24.120252+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerClientBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)> + IOAudioEngineUserClient[<private>]::registerClientBuffer64  (<private>[1], 0x1019f5000, 0x8010, 0x6a)
2020-12-27 00:01:24.120253+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::findBufferSet(unsigned int)>   null clientBufferSetList
2020-12-27 00:01:24.120254+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::findBufferSet(unsigned int)>   did not find clientBufferSetList for ID 0x6a
2020-12-27 00:01:24.120302+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerClientBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)>   clientBuffer->mAudioClientBuffer32.sourceBuffer before offset: <private>, offset size: 16
2020-12-27 00:01:24.120304+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerClientBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)>   clientBuffer->mAudioClientBuffer32.sourceBuffer after offset: <private>
2020-12-27 00:01:24.120305+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::findBufferSet(unsigned int)>   null clientBufferSetList
2020-12-27 00:01:24.120306+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::findBufferSet(unsigned int)>   did not find clientBufferSetList for ID 0x6a
2020-12-27 00:01:24.120306+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerClientBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)>   creating new IOAudioClientBufferSet
2020-12-27 00:01:24.120308+0800  localhost kernel[0]: (IOAudioFamily) <IOAudioFamily`IOAudioEngineUserClient::registerClientBuffer64(IOAudioStream*, unsigned long long, unsigned int, unsigned int)>   output
[... cut ...] 0: Flipping display with wsaa 1 flipped from 1920 x 1080 -> 1920 x 1080 on 4137a000 with 801 options at stride 7680
2020-12-27 00:01:24.210471+0800  localhost kernel[0]: (AppleIntelKBLGraphicsFramebuffer) <AppleIntelKBLGraphicsFramebuffer`AppleIntelPlane::configurePlane(FlipTransactionArgs*)> [IGFB][LOG  ][PLANE     ] [Flip_Transaction] Pipe Gamma Enable
2020-12-27 00:01:24.227659+0800  localhost kernel[0]: (AppleRTC) <AppleRTC`AppleRTC::setGMTTimeOfDay(long)> RTC: setGMTTimeOfDay 1608998487
```

Dievale, 2020-12-26:
Same issue closing the lid.

archfish, 2020-12-27:
I figure it out, it cause by `EFI/OC/ACPI/SSDT-INPUT.aml`, adb mapping was missing, fixed by:

```aml
/*
 * Intel ACPI Component Architecture
 * AML/ASL+ Disassembler version 20200528 (64-bit version)
 * Copyright (c) 2000 - 2020 Intel Corporation
 * 
 * Disassembling to symbolic ASL+ operators
 *
 * Disassembly of iASLpmeJn2.aml, Sun Dec 27 11:45:49 2020
 *
 * Original Table Header:
 *     Signature        "SSDT"
 *     Length           0x0000010A (266)
 *     Revision         0x02
 *     Checksum         0x09
 *     OEM ID           "hack"
 *     OEM Table ID     "INPUT"
 *     OEM Revision     0x00000000 (0)
 *     Compiler ID      "INTL"
 *     Compiler Version 0x20200528 (538969384)
 */
DefinitionBlock ("", "SSDT", 2, "hack", "INPUT", 0x00000000)
{
    External (_SB_.PCI0.LPCB.PS2K, DeviceObj)

    Scope (\_SB.PCI0.LPCB.PS2K)
    {
        Name (RMCF, Package (0x02)
        {
            "Keyboard", 
            Package (0x04)
            {
                "Swap command and option", 
                ">y", 
                "Custom ADB Map", 
                Package (0x03)
                {
                    Package (0x00){}, 
                    "e001=34", 
                    "e01d=3d"
                }
            }
        })
    }

    Scope (\_SB.PCI0.LPCB.PS2K)
    {
        Method (_DSM, 4, NotSerialized)  // _DSM: Device-Specific Method
        {
            If (!Arg2)
            {
                Return (Buffer (One)
                {
                     0x03                                             // .
                })
            }

            Return (Package (0x04)
            {
                "RM,oem-id", 
                "HPQOEM", 
                "RM,oem-table-id", 
                "ProBook-87"
            })
        }
    }
}
```

----- 

But I meet other issue now : itlwm can not connect wifi - -||

I decide rolleback my OC .

Joaotcs, 2020-12-27:
> Use my tpl0.aml acpi patch and add it on config.plist
> 
> Yes, I also notice this bug with closing lid, it gets a sound like I am constantly pressing a Key... It is from newer commits. Maybe the input.aml cleaning did this. Will try the older version acpi patch.

I suspected that @archfish  :p Thanks for the finding and posting it ;), with Christmas and other stuff, i had no time to investigate.

About ITLWM, I am not certain what triggered to not work, is it enough to rollback OC or you have to change ACPI patches also?

BTW, a native card is always better than that...
Regards

Joaotcs, 2020-12-27:
I have updated my Github,  with the fix guys.
Any error, please discuss there, as @kecinzer will leave the scene, at least as an active user.

Regards

PS I am not a developer, but I will at least keep it updated and if I find any "fix" will post.

Ermac-X, 2020-12-27:
I will download and try it out hope it will work

Joaotcs, 2020-12-27:
Maybe ITLWM doesnt work because USB mapping, try mine... If it doesnt work, try mine with the next modification:
In config.plist disable USBPort.kext and enable XHCIPortLimit. Test if it works this way, if it does, you have to map your USB ;)

Ermac-X, 2020-12-27:
The fix works for sleep now system sleeps with AC plugged without any Problem 👍 
my touchpad and touch still persistence though and now no screen light control, this is weird but hard for me to trace and fix since i'm not familiar with all the things you do  .. I simply used your Bootloader from the beginning as those are my capabilities in Hackintosh .

Joaotcs, 2020-12-27:
you are right, i will adress that, and post here the fix soon.
thanks

Joaotcs, 2020-12-27:
All fixed now ;)

Joaotcs, 2020-12-27:
About touchpad and screen, do you have windows?
Please dump DSDT and tell me the ID devices for touchpad and touchscreen...
Regards

Ermac-X, 2020-12-27:
first of all thank you very much for your effort to help i hope i got what u want correctly 
so this is my DSDT dump from Windows and the touchpad and screen id if u got it right 
**HID-compliant touch screen**
HID\VEN_ELAN&DEV_25D0&Col01
HID\ELAN25D0&Col01
HID\*ELAN25D0&Col01
HID\VID_04F3&UP:000D_U:0004
HID_DEVICE_UP:000D_U:0004
HID_DEVICE
**HID-compliant mouse**
HID\VEN_ALP&DEV_0017&Col01
HID\ALP0017&Col01
HID\*ALP0017&Col01
HID\VID_044E&UP:0001_U:0002
HID_DEVICE_SYSTEM_MOUSE
HID_DEVICE_UP:0001_U:0002
HID_DEVICE

[DSDT.zip](https://github.com/kecinzer/hpelitebook850g5-opencore/files/5744687/DSDT.zip)

Joaotcs, 2020-12-27:
Hello sir! It should be working now.
I forgot to add a kext to my repo!
I was looking in the wrong place !

Ermac-X, 2020-12-27:
thank you very much ! it is now working fully touch trackpad and sleep perfectly.
- wonder if I can have dual boot on OC like old fellow clover ? my googling revealed it is only working with two SSDs not one like our case ... any ideas ? 
- I guess my only persistent problem with this device is the webcam like some of you ... it is seen by the system but doesn't 
- work ... 
as i understood its a matter of luck with Mac either it does work out of the box or forget no way to solve its problem 
before Covid it was not important feature but now very necessary.
sure can solve simply with USB webcam, yet not handy to walk with it everywhere... 
anyway thats not a crucial issue now i again have a fully working system with all the basic functions ! 
thank you again !

mehmetyukcell, 2020-12-27:
BATT.zip

Please, should be tried using ACPI file and patch with OpenCore.

mehmetyukcell, 2020-12-27:
https://github.com/KirillSerogodsky/HP-ProBook-430-440-450-G6-Hackintosh/files/5744814/BATT.zip

Ermac-X, 2020-12-27:
Not working touchpad and touch screen after sleep problem still present.

Joaotcs, 2020-12-27:
see if you need to fix hibernation on hackintool.

Ermac-X, 2020-12-28:
I already checked it is fixed .. maybe sleep issue was one time random thing

ziyilang2000, 2020-12-28:
use https://github.com/kecinzer/hpelitebook850g5-opencore/issues/48#issuecomment-751422491 Hotkeys cannot adjust brightness and volume
Inspired by https://github.com/kecinzer/hpelitebook850g5-opencore/issues/48#issuecomment-751422491
I quoted [kinoute's](https://github.com/kinoute/Hack-HP-EliteBook-850-G5) configuration，I used to use it.  fixing this bug, I feel that he works perfectly.

>```
> DefinitionBlock ("", "SSDT", 2, "hack", "INPUT", 0x00000000)
> {
>     External (_SB_.PCI0.LPCB.PS2K, DeviceObj)    // (from opcode)
> 
>     Scope (\_SB.PCI0.LPCB.PS2K)
>     {
>         Name (RMCF, Package (0x04)
>         {
>             "Synaptics TouchPad", 
>             Package (0x02)
>             {
>                 "DisableDevice", 
>                 ">y"
>             }, 
> 
>             "Keyboard", 
>             Package (0x04)
>             {
>                 "Function Keys Special", 
>                 Package (0x0F)
>                 {
>                     Package (0x00){}, 
>                     "3b=3b", 
>                     "3e=3e", 
>                     "3f=3f", 
>                     "40=40", 
>                     "42=42", 
>                     "43=43", 
>                     "44=44", 
>                     "e05f=e05f", 
>                     "e06e=e06e", 
>                     "e012=e012", 
>                     "e017=e017", 
>                     "e02e=e02e", 
>                     "e030=e030", 
>                     "e001=e001"
>                 }, 
> 
>                 "Function Keys Standard", 
>                 Package (0x0F)
>                 {
>                     Package (0x00){}, 
>                     "3b=e05f", 
>                     "3e=e06e", 
>                     "3f=e012", 
>                     "40=e017", 
>                     "42=e02e", 
>                     "43=e030", 
>                     "44=e001", 
>                     "e05f=3b", 
>                     "e06e=3e", 
>                     "e012=3f", 
>                     "e017=40", 
>                     "e02e=42", 
>                     "e030=43", 
>                     "e001=44"
>                 }
>             }
>         })
>     }
> 
>     Scope (\_SB.PCI0.LPCB.PS2K)
>     {
>         Method (_DSM, 4, NotSerialized)  // _DSM: Device-Specific Method
>         {
>             If (LNot (Arg2))
>             {
>                 Return (Buffer (One)
>                 {
>                      0x03                                           
>                 })
>             }
> 
>             Return (Package (0x04)
>             {
>                 "RM,oem-id", 
>                 "HPQOEM", 
>                 "RM,oem-table-id", 
>                 "ProBook-87"
>             })
>         }
>     }
> }
> 
>```

Joaotcs, 2020-12-28:
Check my repo, I already had that fixed m8 ;)
Plus other modifications

Yes, you can have dual boot with opencore, I have it...
Just make sure you save your EFI, you will need it after again ;)
( please check one issue on kecinzer repo, thats closed now but i teached how to do the dual boot to another user).

Ermac-X, 2020-12-28:
I checked .. The files are still from 16 hours  , have you uploaded anything new just right now, or you mean yesterday modifications ? 
Cause with yesterday update the touchpad and screen freeze problem was still there after a couple of sleeps. 

Concerning Dual boot it seems i have to read some closed issues to find that one but maybe by that learn something new :D

Joaotcs, 2020-12-28:
https://github.com/Joaotcs/Hackintosh-Elitebook-8x0-g5/issues/4

I was talking about keyboard.. 
Well, I haven't had any sleep issues now. But only time will tell. I have no ideas for new fixes at the moment...
But this is near perfect now

Ermac-X, 2020-12-29:
@kecinzer 
sleeping with AC is working now 
touchpad freeze issue  after some several sleeps is also solved 
keyboard function keys brightness and volume are not working with the latest fix 
something i noticed now after sleeping many time, the device starts to  sleep and wake immediately without any program Interfering. 
another notice sleep takes longer time in comparison with old builds

archfish, 2021-01-03:
> Maybe ITLWM doesnt work because USB mapping, try mine... If it doesnt work, try mine with the next modification:
> In config.plist disable USBPort.kext and enable XHCIPortLimit. Test if it works this way, if it does, you have to map your USB ;)

I have rollback my OC for a week, work fine. ITLWM not working maybe related to keybord map, E.g wifi switch set to false. When using my old OC it cause same problem.
I will keep using old OC before I have time to test.