kecinzer/hpelitebook850g5-opencore #49: Switch to Apple HW

kecinzer, 2020-12-21:
I have bad new here for all followers of this repo.
Few days ago I finally bought real Mac. It's new MacBook Air M1 and I must say, that using this machine is the best user experience that I had with any Apple product.
Right now I working only on new MBA and deciding what to do with hackintosh. I spent great time with it. My machine works almost great. But, it consumes very much of time. And because of my nature I spent even more time investigating issues and problems, that start to be annoying for me (hibernation, low battery life, heat, etc.).
When Apple announced Apple Silicon I was very excited. I only considering wait for second generation or buy it right now. And because I'm early adopter and price start to make sense for me, I bought MacBook Air.

Elitebook is my company machine and current models are based od AMD, so hackintosh will not be possible in future for me without buying my own machine.
I don't want to definitely close this project, because, I don't know when I get the mood for hackintosh again. For now make more sense for me to install Windows 10 on Elitebook.

I hope that some one who has fork this repo (@Joaotcs) will continue with updates and more features.

Joaotcs, 2020-12-21:
Thank you for everything!
From what I have seen, m1 is marvillous. Ofc it's 1st gen, but 2nd gen will probably increase 50% price!

Best of luck 😉

Rixxx-git, 2020-12-22:
Thanks a lot for everything @kecinzer !

Can you please describe what you like the most about MB Air with M1? I am thinking of buying it as well. Is it as good as the Elitebook with 32GB RAM and 8th generation i5?

archfish, 2020-12-22:
Thanks a lot! Best of luck :)

igorkolesnikov13, 2020-12-22:
Hey, @kecinzer . Really bad news. Was following your repo for about a year and it was super fine work.
I changed to this repos after using RehabMan's work (https://www.tonymacx86.com/threads/guide-hp-probook-elitebook-zbook-using-clover-uefi-hotpatch.261719/).

Thanks a lot for your work and best of luck!

kecinzer, 2020-12-22:
> Thanks a lot for everything @kecinzer !
> 
> Can you please describe what you like the most about MB Air with M1? I am thinking of buying it as well. Is it as good as the Elitebook with 32GB RAM and 8th generation i5?

I like almost everything. Better form factor than HP. It's incredible fast, battery stands for around 18 hours. There are almost no heat at all. It's still cold and don't have any fan. Retina display is nice for use (but I had it on MB Pro 15" 2014). Keyboard is great with touchID, touchpad with force touch is magical :). Speakers can't be compared to my Elitebook plays like on external speakers. Everything works OOB and you don't need to play with it like on hackintosh :).
I was scared about maximum of 16GB RAM, as I have 32 GB on Elitebook, but finally I take 8GB RAM version and it's enough for me. I din't notice any issues with performance so far. I think that Apple finally release model that has great price value.

Only cons for me is support for only one external display. There is workaround with DisplayLink, but only for office work.
I used to 13" size very quickly. It's super light and mobile.

Rixxx-git, 2020-12-23:
Thank you for your insights!

I switched from my Macbook Pro to Hackintosh a few years ago, because I love MacOS, but I hated not being able to replace any part of the Macbook. The retina display of two of my Macbook Pro models had the #staingate problem. Apple only has a replacement program for that problem for models not older than a few years. Which I think is ridiculous. So I thought what happens if the SDD, keyboard or any other part has a problem. Not to mention that you can't upgrade the RAM later on.

At the end you will be forced to buy a new Macbook every 3 to 5 years if you need a good performance for work or in case any part doesn't work anymore. 

I bet the new M1 is amazing. But I will stay with my 830 g5 as long as possible. 🙂

kecinzer, 2020-12-23:
@Rixxx-git that's funny, I have same experiences. I switched my MBP 13" display once and 15" almost twice - second time I was out of staingate date :). I completely understand you, but that's Apple. That's why I really like my Elitebook. I can bought high speed SSD and more RAM.
But now with Apple Silicon I think that Hackintosh gonna disappear by the time.
And price of new MB Air is acceptable for me even for 3 year lifecycle.

Rixxx-git, 2021-02-11:
Hi @kecinzer, I might switch to a macbook m1 as well. How do I uninstall mac and install only windows 10 on my HP Laptop? 🙂

kecinzer, 2021-02-12:
> Hi @kecinzer, I might switch to a macbook m1 as well. How do I uninstall mac and install only windows 10 on my HP Laptop? 🙂

Hi, just make Windows 10 installation USB and install windows. You can wipe you disk durring installation. Also you should change your BIOS settings (may be reset to defaults).

Rixxx-git, 2021-02-12:
Done, HP sold and writing on my Macbook Air M1 now. :) thanks a lot for all your hackintosh work but I don't think I will ever get back to it. All the best for you guys!