kecinzer/hpelitebook850g5-opencore #50: Kernel panic when booting installer on 840 G5

otaviomad, 2021-01-29:
While trying to boot on my 840 G5 I get an nvme related kernel panic. I've attached a picture of the full logs. I'm using the MacBookPro15,4 SMBIOS.
[image]

otaviomad, 2021-01-29:
I've attached a picture of my system specs as declared by speccy

[image]

archfish, 2021-01-31:
Replace your ssd device, MTFDHBAXXXTCK was not supported.

otaviomad, 2021-02-02:
wow, that's unfortunate. thanks for the reply.

closed 2021-02-02 by otaviomad

ammi3, 2021-04-08:
> While trying to boot on my 840 G5 I get an nvme related kernel panic. I've attached a picture of the full logs. I'm using the MacBookPro15,4 SMBIOS.
> [image]
So, can it work now. My notebook is the same as yours，can you tell me the version of the system you installed and the EFI used. Thank you.

otaviomad, 2021-04-08:
> So, can it work now. My notebook is the same as yours，can you tell me the version of the system you installed and the EFI used. Thank you.

No, and it probably never will. I just downloaded the EFI from this repo and was trying to install big sur.