kecinzer/hpelitebook850g5-opencore #53: Kernel panic on install

mahrks, 2021-05-26:
Followed instructions fully, SMBIOS generated and config replaced, did BIOS instructions as much as possible with a successful boot. A bit more than halfway through the install, everything freezes up, mouse cannot move, usb power light on static. Doesn't install at all.
Specs:
HP EliteBook 840 G5 @ 120hz w/ Sure view privacy
i5-8250U UHD 620
16GB DDR4
SAMSUNG internal NVME 256GB

P.S. I think my model is different than traditional EliteBooks, as the Remote Management options wasn't in BIOS>Advanced at all. The last EliteBook 840 G5 I owned macOS worked great on. Same BIOS version regardless.

Thanks in advance

labrouss, 2021-05-26:
> I think it's my SSD, as people have had kernel panic issues in the past. I'm gonna try updating the firmware with Samsung software and see if it works.

Hi,

If you have a PM981, the moment the installer will be accessing the NVME disk it will cause a kernel freeze. 

You can try the NVMEfix.kext 

https://github.com/acidanthera/NVMeFix

but I think you are safer by buying one that is of the self supported on OSX.

mahrks, 2021-05-26:
tried it, now it goes a bit further. freezes exactly at 14 minutes in for big sur
i don't have another ssd at the moment but users have said "update the firmware," don't know how to

labrouss, 2021-05-26:
So far I don’t know anyone that has the PM981 and managed to succeed.

mahrks, 2021-05-26:
me, on my old 840 g5 with clover actually. this one doesnt support clover
though (instant reboot upon selecting macos)